analogdevicesinc/PyTrinamic #2: What is the status of this module?

synatree, 2019-03-05:
Is this module ready for use, or still in active development?

trinamic-ed, 2019-03-05:
We started with the base communication with USB-CDC and UART. We will add the full IC support like in the "TMC-API" project and add support for the evaluation boards comparable with the "TMC-EvalSystem" firmware project. Some standard modules will also follow, before we link this project officially on our website.

closed 2019-05-10 by trinamic-LH

synatree, 2019-05-17:
Thanks, we're successfully using what you have already btw.

trinamic-LH, 2019-05-20:
Good to hear!

Any specific ICs/Evals/Modules you're missing in the PyTrinamic package right now? We're working on getting most of them integrated into the package - if you need a specific one now/in the near future, let us know :)