analogdevicesinc/PyTrinamic #3: ModuleNotFoundError: No module named 'serial'

dsitnik, 2019-07-11:
[System: Fedora 29, Python 3.7]
Hi,

after running $ sudo python3 setup.py install I got:

Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from PyTrinamic import __version__
  File "/home/dsitnik/PyTrinamic/PyTrinamic/__init__.py", line 8, in <module>
    from PyTrinamic.connections.serial_tmcl_interface import serial_tmcl_interface
  File "/home/dsitnik/PyTrinamic/PyTrinamic/connections/serial_tmcl_interface.py", line 7, in <module>
    from serial import Serial, SerialException
ModuleNotFoundError: No module named 'serial'

I get the same error after installing pytrinamic via `$pip3 install pytrinamic`
I tried installing the module serial using `$pip3 install serial` but after that I got the following error:
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from PyTrinamic import __version__
  File "/home/dsitnik/PyTrinamic/PyTrinamic/__init__.py", line 8, in <module>
    from PyTrinamic.connections.serial_tmcl_interface import serial_tmcl_interface
  File "/home/dsitnik/PyTrinamic/PyTrinamic/connections/serial_tmcl_interface.py", line 7, in <module>
    from serial import Serial, SerialException
ImportError: cannot import name 'Serial' from 'serial' (/usr/local/lib/python3.7/site-packages/serial/__init__.py)

Any ideas what went wrong? Thank you in advance.

Lionheart1810, 2019-07-11:
Normally, the module serial from the package pyserial should get installed automatically on PyTrinamic setup. I will take a look into that.
What you could try is running `pip install pyserial` and `pip install python-can` manually. If you want to run the tests, you will need `pip install openhtf` aswell.

Lionheart1810, 2019-07-12:
`pyserial` has been added to the requirements and should now be installed automatically.
I've tested it and it worked accordingly. Feel free to reopen the issue if this problem persists.

closed 2019-07-12 by Lionheart1810

TakingAway, 2019-12-11:
`sudo pip3 install pyseria` helped for me, not `pip install pyserial`.