analogdevicesinc/PyTrinamic #4: SPI support

servetcoskun, 2019-10-28:
Is there a timeline or any plans on SPI support?

trinamic-LH, 2019-10-29:
Right now we only support TMCL over Serial, USB and CAN.

Support for UART directly to an IC is built into the examples of the TMC4671 but still requires some work to become general-purpose for other ICs.

Support for the RTMI (SPI Slave connection to a TMC4671) is planned, but hasn't been started yet.

We haven't considered a direct connection to an IC via SPI yet. If i am not misunderstanding you here - is that what you intend to use?

dzid26, 2019-11-15:
I am trying to figure it out too. 
 
SPI is related to ICs and it would make perfect sense to support it if ICs are to be supported. 
But it seems as you said, mainly TMCL protocol is supported here - which means modules.

I think it would be helpful to fill out Readme.md with details as to what this library does and what it is intended to do in the future. 

Thanks

PMTaylor, 2019-12-28:
Agreed! SPI soon please? I'm hacking together something with your register + field enumeration files, but would love to have it done well by folks who understand the chip better.

I'd love to use Python to control the TMC4361A BOB on a Raspberry Pi. SPI connection is the move in that case, right? 
I do see an example on your blog in C, but i'd much prefer to use python for this.

trinamic-LH, 2020-02-03:
In order for SPI support to work properly, all the register logic [similar to the Evalsystem](https://github.com/trinamic/TMC-API/blob/86ea27d9b968eb53a90d1f5baa0ab9127c950b51/tmc/ic/TMC5130/TMC5130.c#L17) would have to be reimplemented [in PyTrinamic](https://github.com/trinamic/PyTrinamic/blob/6fa380af29bcee57782616ad04705a819c32d0bb/PyTrinamic/ic/TMC5130/TMC5130.py#L28) This includes access permissions, caching via shadow registers and proper initialization.

Currently there are no plan for us to implement this in the PyTrinamic library. Our primary intended use-case for this library is to enable scripting of the tasks previously only doable manually in the TMCL IDE. Raw, direct IC access is not part of this task.

closed 2020-02-03 by trinamic-LH

Neex101, 2020-03-30:
Would be very good to have Python/SPI hooks into the chip - would make evaluation for noobies much more accessible...