analogdevicesinc/PyTrinamic #5: Modules

dzid26, 2019-11-17:
Are you planning to add TMCM-1276 module in the near future?
It is currently a candidate for my design.

trinamic-jm, 2019-11-26:
Hello,

we are currently working on implementing this module to the PyTrinamic project.

trinamic-jm, 2019-11-26:
The TMCM-1276 has been implemented. If you have any questions left or trouble with the program in the near future, please let us know.

dzid26, 2019-12-01:
Thanks a lot.
Is it possible to also add TMCM-1270 support?

trinamic-LH, 2019-12-02:
Yes

We plan on adding all ICs, Evals and Modules over time. By asking for specific ones we can do those first.

trinamic-jm, 2019-12-05:
The TMCM-1270 has also been implemented. If you have any questions left or trouble with the program in the near future, please let us know. Also if you have any questions to other Chips feel free to ask us.

closed 2019-12-05 by trinamic-jm