analogdevicesinc/PyTrinamic #6: TMCL-1270 and ESD CAN-USB/2 support

Zebrafish007, 2019-12-03:
Will there be TMCM-1270 support? I am using ESD CAN-USB/2 device and somehow I can't get it working with NTCAN provided by ESD.

trinamic-jm, 2019-12-18:
We Implemented the TMCM-1270.

As for the ESC CAN adapter - we do not use this adapter and it seems that the library we are using for CAN does not support it either [1].

Right now we only have PCAN as a CAN adapter implemented and will also add the Kvaser CAN adapter. If you have one of those available to use we recommend switching to those.

If you aren't able to switch to one of those supported adapters we could assist you in implementing an adapter implementation into this project - the integration into the ConnectionManager and TMCL to be specific.

[1]: https://python-can.readthedocs.io/en/master/interfaces.html

Zebrafish007, 2019-12-19:
Thanks, looking into it as we speak. I've also contacted ESD and forwarded them the URL to this discussion. 

I am trying to update/install latest version via pip but fails because PyTrinamic/PyTrinamic/version.py / is still version 0.1.16. Uninstall and reinstall don't make a difference. Not sure why.. but I guess its cache related although the .whl was removed from folders. Any change to update it to 0.1.16.1 or something?

Zebrafish007, 2019-12-19:
with `pip install pytrinamic --no-cache-dir` its getting the new file but its not updated.

trinamic-LH, 2019-12-19:
We have not released a new version in a while - 0.1.16 is still the newest release. We'll create a new release in the coming days.

Zebrafish007, 2019-12-19:
> could assist you in implementing an adapter implementation into this project - the integration into the ConnectionManager and TMCL to be specific.

This would be wonderfull because changing adaptor is costly for me at this stage. I have from ESD the python NTCAN library (PyNTCAN-2.2.0-py3.7). How to proceed?

trinamic-LH, 2019-12-19:
You will need to create a new class in PyTrinamic/connections/
Since we will send TMCL over the CAN interface, the class should inherit from the `tmcl_interface` class. Your class needs to [implement _send() and _recv() functions](https://github.com/trinamic/PyTrinamic/blob/1733df989346cfd091015c1bae56a8254439f234/PyTrinamic/connections/tmcl_interface.py#L13) in order for the TMCL interface class to work.

Once the above is implemented you can connect to a module using the new interface by [hardcoding the interface connection](https://github.com/trinamic/PyTrinamic/blob/e68bf4344f612be97a0ffde64cdc6fd7f5810879/PyTrinamic/examples/evalboards/TMC5041/TMC5041_eval_register_dump.py#L17) (link is of an older code version). This allows you to test your implementation, but it is not recommended to use long-term. Instead your class should be intergrated into the [ConnectionManager](https://github.com/trinamic/PyTrinamic/blob/1733df989346cfd091015c1bae56a8254439f234/PyTrinamic/connections/ConnectionManager.py#L18) to allow flexible connection selection via the command line.

Your interface class will also need a close() and a list() function to be compatible with the ConnectionManager. (A supportsTMCL() function is also needed, but already provided by the tmcl_interface class.) Once all required functions are added your interface can be [integrated into the ConnectionManager](https://github.com/trinamic/PyTrinamic/blob/1733df989346cfd091015c1bae56a8254439f234/PyTrinamic/connections/ConnectionManager.py#L79). 

Also add the library dependency to [setup.py](https://github.com/trinamic/PyTrinamic/blob/1733df989346cfd091015c1bae56a8254439f234/setup.py#L22).

trinamic-jm, 2019-12-20:
We released our new version - [0.1.17 in PyPI](https://pypi.org/project/PyTrinamic/0.1.17/#history)  
This version contains all further updates since October.  If you have any questions left please let us know.

Zebrafish007, 2019-12-20:
`pip search pytrinamic` shows 0.1.16.  With `pip search pytrinamic=0.1.17` pip installs the latest version.
Checked content of folders and does not corresponds with newly added content. Missing folder "examples" under root/anac/lib/site-packages/pytrinamic. It just dumps them in "root/anac/scripts" which i.m.o. messes up that folder, and in modules TMCM_1270/1636.py and in ic/TMC2209 aren't there.

trinamic-LH, 2019-12-20:
The pip version sounds like a caching issue.

The examples are marked as scripts, not as modules - which dumps them all in the scripts/ folder. This is the standard behaviour of the python package system, we aren't happy with that either. As of right now we do not have a better approach to delivering examples for all the different ICs/Evals/Modules.

Zebrafish007, 2019-12-20:
I assume on pip's end its a caching issue? Just tried the `--no-cache-dir` option after uninstall.

/examples : if you try "py_examples" or "docs_&_exmaples" as folder, would that work?

Zebrafish007, 2019-12-20:
..or "test_examples"?

trinamic-LH, 2019-12-20:
> I assume on pip's end its a caching issue? Just tried the `--no-cache-dir` option after uninstall.
Probably, i tried that too and got 0.1.16. Specifying PyTrinamic==0.1.17 during the install command worked.
 
> /examples : if you try "py_examples" or "docs_&_exmaples" as folder, would that work?
It has been a bit since i looked into the python packaging tools/mechanisms. As far as i remember there is no direct way of telling Python to put the files into a different subfolder. Even the location of the scripts/ folder wasn't officially written down (or i didnt find that). One idea would be to provide the scripts zipped as a download or something.

trinamic-jm, 2019-12-20:
We fixed the setup.py and added the missing file-paths. It should be a [new Version in PyPi](https://pypi.org/project/PyTrinamic/0.1.18/#history) now avaible. 
Thanks you for the quick error responding. 
We appreciate every support to create the perfect enviroment for our user.

trinamic-sw, 2020-01-06:
We just added SocketCAN support, which also includes the ESD CAN-USB/2 adapter. However SocketCAN is only supported on linux systems.

closed 2020-01-31 by trinamic-jm