analogdevicesinc/PyTrinamic #9: TMCM-1640 questions

dsitnik, 2020-01-29:
Hi,

I have two questions. First, is there a way to detect when the motor reached hardstop? Currently, I am doing it by detecting if two consecutive positions are the same.

Second, if I want to move the motor positions manually (by hand), can I disable motor phases (if that is the thing that is blocking manual movements) but keep reading positions? 

I am using Faulhaber motor 2232S012BX4 with Hall sensor.

Thank you in advance!

trinamic-ed, 2020-01-31:
Hi,

when the motor reached the hardstop in velocity or position mode, you can see the actual current rising to the max current.

I just added enable/disable driver parameter (AP 255) to the next firmware version v2.10 which will be released in the next days. I can provide you an preliminary version here: 

https://www.dropbox.com/s/pgelcwnwcd8t9vo/TMCM-1640_TMCL_Hw1.0_Fw2.10.hex?dl=0

Best regards,
      Enrico

dsitnik, 2020-01-31:
Thank you for the preliminary version. I really appreciate it. Regarding the first question, I cannot find a parameter nor a function that can return the actual current or max current. What function should I call to get these values?

trinamic-ed, 2020-01-31:
AP 150 actual current
AP 6 max current

dsitnik, 2020-01-31:
Excellent, thank you very much!

dsitnik, 2020-02-01:
I tried updating firmware on my module. I get an error:

```
Firmwareupdate started @ttyACM0/USB/id1/TMCM-1640
Active firmware: 1640V209
Enter bootloader mode...
Disconnecting...

An error has occurred
```
After I restart my module I still have an old firmware version. I tried on Windows and on Linux with both 3.0.24 and 3.0.25 TMCL-IDE versions.

trinamic-ed, 2020-02-03:
Hi,

please try to start the firmware update with the IDE, when the error occurs, close the IDE, open the IDE and start again. Do not repower the module. 

Regards,
     Enrico

dsitnik, 2020-02-04:
If I don't repower the module, it cannot be seen from the IDE when I restart the IDE and try to update the firmware.
[https://ibb.co/XYM72RF](url)

dsitnik, 2020-02-08:
I just found out that your advice works on Windows, but not on Linux IDE

closed 2020-02-08 by dsitnik