analogdevicesinc/PyTrinamic #10: AttributeError: module 'PyTrinamic' has no attribute 'showAvailableComPorts'

dsitnik, 2020-02-01:
version = 

```
import PyTrinamic
PyTrinamic.showInfo()
PyTrinamic.showAvailableComPorts(USB=True)
```
Your example code is not working. I get an error:
AttributeError: module 'PyTrinamic' has no attribute 'showAvailableComPorts'

version=0.1.18

dsitnik, 2020-02-01:
Also in line:
https://github.com/trinamic/PyTrinamic/blob/6fa380af29bcee57782616ad04705a819c32d0bb/PyTrinamic/examples/modules/TMCM_1640/TMCM_1640_hall_positioning_test.py#L48
AttributeError: 'TMCM_1640' object has no attribute 'COMM_MODE_FOC_HALL'
Should it be:
`module.setCommutationMode(module.GPs.COMM_MODE_FOC_HALL)`?

Also:
https://github.com/trinamic/PyTrinamic/blob/6fa380af29bcee57782616ad04705a819c32d0bb/PyTrinamic/examples/modules/TMCM_1640/TMCM_1640_hall_positioning_test.py#L61
`Traceback (most recent call last):
  File "motor_test.py", line 61, in <module>
    while not module.positionReached():
  File "/usr/local/lib/python3.6/dist-packages/PyTrinamic/modules/TMCM_1640.py", line 94, in positionReached
    return ((self.statusFlags() & self.FLAG_POSITION_END) != 0)
AttributeError: 'TMCM_1640' object has no attribute 'FLAG_POSITION_END'
`

trinamic-jm, 2020-02-03:
Hi,
the 'showAvaibleComPorts' issue is an outdated function from our connection logic. 

As for the AttributeError - we moved axis parameters, global parameters and constants around and didn't adjust this entry in the process. In the just released version 0.01.19 we fixed the breakage you found in this script. However, we will move the COMM_MODE_FOC_HALL - since it is neither an axis parameter nor a global parameter, it is a constant enum value.

closed 2020-02-03 by trinamic-jm