analogdevicesinc/PyTrinamic #13: Is this useful for BOB boards?

Neex101, 2020-03-28:
Hi,
I am new to Trinamic products, so forgive me if this is really obvious.
I have a TMC2660-BOB.
I want to build a test UI using python on a RasPi. So, comms would be from the RasPi SPI pins to the BOB pins.
I'm hoping these libs are what I need. If so, would you be able to document any example implementations in the github wiki?
Many thanks,
Ed

trinamic-jm, 2020-03-30:
Hi Neex101,

as already discussed [here](https://github.com/trinamic/PyTrinamic/issues/4), direct SPI communication with ICs is not part of PyTrinamic.

For your project you should use our low level [TMC-API](https://github.com/trinamic/TMC-API) written in C.

If you have any further questions please let us know.

JM

Neex101, 2020-03-30:
Hi, thanks for your response.
Good to know.
Do you have any guides on how to implement the TMC-API? (its README doesn't tell me much and there's nothing in the wiki).
Best, Ed

trinamic-ed, 2020-11-11:
Hi,

the TMC-API is directly used by the https://github.com/trinamic/TMC-EvalSystem project.

Regards,
     ED

closed 2020-11-11 by trinamic-ed