analogdevicesinc/PyTrinamic #17: Go in negative position

thibww67, 2020-07-01:
I would like to use this library with my TMCM_1161, and by using "moveTo" I can set a positive value but not a negative value, for example :

Module_1161.moveTo(25600) //work 
time.sleep(3)

Module_1161.moveTo(-25600) //doesn't work
time.sleep(3)

Is it possible ?

Thank you

trinamic-LH, 2020-07-01:
Hello,

I tried both positive and negative movement in a dummy program [1] and it looks like the correct values are getting passed to the TMCM1161 [2].

I do not have the means to test this with the module myself right now - i will forward this to the right people internally.

In the meantime, have you tried tweaking your values - for example changing the delays or doing the movement in negative direction before the movement in positive direction?

[1]: My test script. The dummy connection emulates a TMCL connection, printing any datagrams sent by the script):
```py
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1161.TMCM_1161 import TMCM_1161

cm = ConnectionManager("--interface dummy_tmcl")
interface = cm.connect(debug_interface=True)
module = TMCM_1161(interface)

print("Moving forwards")
module.moveTo(25600)
print()

print("Moving backwards")
module.moveTo(-25600)
print()
```

[2]: My test log (with comments)
```
Opened dummy TMCL interface on port 'dummy'
	Data rate:  0
	Host ID:    2
	Module ID:  1
Moving forwards
TMCL_Request: 01,05,00,00,00006400,6A  <- Set axis parameter 0 (target position) to a value of 25600 
TMCL_Reply:   00,00,00,00,00000000,00
TMCL_Request: 01,05,8A,00,00000000,90
TMCL_Reply:   00,00,00,00,00000000,00

Moving backwards
TMCL_Request: 01,05,00,00,FFFF9C00,A0  <- Set axis parameter 0 (target position) to a value of -25600 
TMCL_Reply:   00,00,00,00,00000000,00
TMCL_Request: 01,05,8A,00,00000000,90
TMCL_Reply:   00,00,00,00,00000000,00
```

thibww67, 2020-07-01:
Thank you for your answer, you're right, in fact the issue was the GetActualPosition in negative value. I expected to get the negative value instead of a non-signed value, so my code didn't react as I wanted.

trinamic-LH, 2020-07-01:
That problem should be properly fixed with a6cce11ee6be374076f93e81e44f3af4bd8c4e33

The TMCM1161 was missing the optional signed parameter for `getAxisParameter()`. This parameter is used to automatically convert to signed values when reading. `TMCM1161.getActualPosition()` (and others) now properly use this parameter.

Sidenote: To prevent this sort of issue in general we plan on adding the relevant information (signedness, bit count) of each axis parameter to the declarations in the `_AP` classes. Any call to getAxisParameter will then have all the info necessary for casting values properly.

closed 2020-07-01 by trinamic-LH