analogdevicesinc/PyTrinamic #22: How do i connect to a motor via the CAN interface?

MrGloriousFast, 2020-08-26:
I have the motor 'TMCM1260'.

I want to use the CAN interface to control the motor with the pytrinamic library.

I have a raspberry pi with the 'MCP2515' CAN transceiver. After adding it to the device tree [like this](https://www.beyondlogic.org/adding-can-controller-area-network-to-the-raspberry-pi/), it is recognized correctly as you can see via the dmesg output:

```
pi@rpi3:~ $ dmesg | grep -i can
    [    4.671295] CAN device driver interface
    [    4.696526] mcp251x spi0.0 can0: MCP2515 successfully initialized.
```

Now i have used the following command to create a network interface:

`pi@rpi3:~ $ ip link set can0 up type can bitrate 500000`

It is correctly listed via ifconfig:

```
pi@rpi3:~ $ ifconfig
can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        RX packets 2482  bytes 201812 (197.0 KiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 1348  bytes 172403 (168.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

However i fail to get a connection via the ConnectionManager.py.

When i run the ConnectionManager.py to test the connections i get the following output:

```
pi@rpi3:~/.local/lib/python3.5/site-packages/PyTrinamic/connections $ python3 ConnectionManager.py 
Verifying interfaces list...

List of interfaces: ['dummy_tmcl', 'pcan_tmcl', 'socketcan_tmcl', 'kvaser_tmcl', 'slcan_tmcl', 'serial_tmcl', 'uart_ic', 'usb_tmcl', 'pcan_CANopen', 'kvaser_CANopen']

Performing test run...

Error: No connections available
Test run complete
```

Now my question is what is the next step to use the PyTrinamic library and get a connection to my motor and be able to control it?

MrGloriousFast, 2020-08-26:
Even through the ConnectionManager.py gave me an error, i figured i try to establish one anyway. I used the following code to try and get a connection but was promted with an error.

The code i used:

```
import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
import time

#raspberry setup for gpio
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BCM)

PyTrinamic.showInfo()

connectionManager = ConnectionManager("--interface pcan_tmcl --data-rate 500000", debug=True)
myInterface = connectionManager.connect()
Module_1260 = TMCM_1260(myInterface)
```

Here is the output with the error at the end:

```
pi@rpi3:~ $ python3 can-connection.py 
PyTrinamic - TRINAMIC's Python Technology Access Package
Splitting string: ['--interface', 'pcan_tmcl', '--data-rate', '500000']
Commandline argument list: ['--interface', 'pcan_tmcl', '--data-rate', '500000']
Parsed commandline arguments: Namespace(data_rate=[500000], exclude=[], host_id=[2], interface=['pcan_tmcl'], module_id=[1], port=['any'])

Connection parameters:
	Interface:  pcan_tmcl_interface
	Port:       any
	Blacklist:  []
	Data rate:  500000
	Host ID:    2
	Module ID:  1

Traceback (most recent call last):
  File "can-connection.py", line 14, in <module>
    myInterface = connectionManager.connect()
  File "/home/pi/.local/lib/python3.5/site-packages/PyTrinamic/connections/ConnectionManager.py", line 262, in connect
    self.__connection = self.__interface(port, self.__data_rate, self.__host_id, self.__module_id, debug=debug_interface)
  File "/home/pi/.local/lib/python3.5/site-packages/PyTrinamic/connections/pcan_tmcl_interface.py", line 56, in __init__
    self.__connection = can.Bus(interface="pcan", channel=self.__channel, bitrate=self.__bitrate)
  File "/home/pi/.local/lib/python3.5/site-packages/can/interface.py", line 127, in __new__
    return cls(channel, *args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/can/interfaces/pcan/pcan.py", line 172, in __init__
    self.m_objPCANBasic = PCANBasic()
  File "/home/pi/.local/lib/python3.5/site-packages/can/interfaces/pcan/basic.py", line 363, in __init__
    self.__m_dllBasic = cdll.LoadLibrary("libpcanbasic.so")
  File "/usr/lib/python3.5/ctypes/__init__.py", line 425, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.5/ctypes/__init__.py", line 347, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libpcanbasic.so: cannot open shared object file: No such file or directory
```

Is this the correct way to establish a connection via the CAN interface? Do i need an additional library? If so where can i find it?

I tried searching for the library but nothing came up:

```
pi@rpi3:~ $ sudo apt search libpcan
Sorting... Done
Full Test Search... Done
pi@rpi3:~ $
```

trinamic-LH, 2020-08-26:
Hello,

When you run the connection manager directly, it will perform a small self-test (verify its interface list) and try to do a connection. When no connection parameters are supplied (via commandline or in the [ConnectionManager class initialization](https://github.com/trinamic/PyTrinamic/blob/5bf28738186a9ad49fd8f313d44cfacb96049a3d/PyTrinamic/connections/ConnectionManager.py#L352)) it will try to establish a connection via the default connection - which is USB. We should probably improve the error message there.

The python can library requires a driver for PCAN interfaces. Your error traceback shows that it is trying to load `libpcanbasic.so` but cannot find it. Refer to the [python can library documentation](https://python-can.readthedocs.io/en/master/interfaces/pcan.html) for steps required.

closed 2020-08-26 by trinamic-LH