analogdevicesinc/PyTrinamic #23: Missing 1240 module

MrGloriousFast, 2020-09-21:
Tried using the 1240 motor via usb but got the following error:

```
Traceback (most recent call last):
  File "motor_test.py", line 17, in <module>
    from PyTrinamic.modules.TMCM1240.TMCM_1240 import TMCM_1240

```

After checking the modules folder i couldn't find the correct module. Assuming its not implemented yet. Is it possible to add it to PyTrinamic?

trinamic-LH, 2020-09-21:
Hello,

yes, the TMC1240 is not implemented yet. We will add this and notify you once it is implemented.

MrGloriousFast, 2020-09-21:
Thanks,

any guesstimate when i can work with it?

trinamic, 2020-09-21:
Hi, we will add the module in the next days. Until then you can use e.g. the TMCM-1260, because the parameter are mostly compatible.

Regards,
     Enrico

trinamic-LH, 2020-09-22:
Hello,

the TMCM1240 has been added in the latest release (v0.1.25)

Regards,
Lenard

closed 2020-09-22 by trinamic-LH