analogdevicesinc/PyTrinamic #25: Module_1260.moveTo() does not work

MrGloriousFast, 2020-09-22:
i tried using the function to move to a specific position. However the motor does nothing and ignores the command.

here is an example code:

```
import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
import time, sys

connectionManager = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000", debug=True)
myInterface = connectionManager.connect()
Module_1260 = TMCM_1260(myInterface)

max_accl = 60000
speed = 10000

try:

  #set motor parameters
  print("Preparing parameters")
  Module_1260.setMaxAcceleration(max_accl)

  #set the starting position as the new zero
  Module_1260.setActualPosition(0)
  print ( 'pos at beginning', Module_1260.getActualPosition())

  #turn for a while
  print("Rotating")
  Module_1260.rotate(speed)
  time.sleep(5);

  #stop
  print("Stopping")
  Module_1260.stop()
  time.sleep(5);
  print ( 'pos after rotating', Module_1260.getActualPosition())

  #turn back to starting position
  print("Rotating back")
  Module_1260.moveTo(0)
  time.sleep(5)
  print ( 'pos after reverting', Module_1260.getActualPosition())

#prevent the motor from spinning when we exit via CTRL+C
except KeyboardInterrupt:
  print('shutting down motor and exit')
  Module_1260.stop()
  myInterface.close()
  quit()

```

And here is the output:

```
$ python3 motor-test.py
Splitting string: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000']
Commandline argument list: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000']
Parsed commandline arguments: Namespace(data_rate=[1000000], exclude=[], host_id=[2], interface=['socketcan_tmcl'], module_id=[1], port=['any'])

Connection parameters:
        Interface:  socketcan_tmcl_interface
        Port:       any
        Blacklist:  []
        Data rate:  1000000
        Host ID:    2
        Module ID:  1

Opened bus on channel can0
Preparing parameters
TMCL_Request: 01,05,05,00,0000EA60,55
TMCL_Reply:   02,01,64,05,0000EA60,00
TMCL_Request: 01,05,01,00,00000000,07
TMCL_Reply:   02,01,64,05,00000000,00
TMCL_Request: 01,06,01,00,00000000,08
TMCL_Reply:   02,01,64,06,00000000,00
pos at beginning 0
Rotating
TMCL_Request: 01,05,02,00,00002710,3F
TMCL_Reply:   02,01,64,05,00002710,00
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,00000000,00
TMCL_Request: 01,06,01,00,00000000,08
TMCL_Reply:   02,01,64,06,0000C38C,00
pos after rotating 50060
Rotating back
TMCL_Request: 01,05,00,00,00000000,06
TMCL_Reply:   02,01,64,05,00000000,00
TMCL_Request: 01,06,01,00,00000000,08
TMCL_Reply:   02,01,64,06,0000C38C,00
pos after reverting 50060

```

As you can see the motor didn't move after the moveTo command.

Also just so you know i will use the 1240 with the same code. should i open an issue for that one too?

trinamic-LH, 2020-09-22:
Hello,

i think i know the issue. I can investigate it later this week.
You do not need to open a separate issue for the TMCM1240, i will check that as well.

Regards,
Lenard

trinamic-LH, 2020-09-25:
Fixed in e6c109fbcde39cbd7e9e5d6bed955560f8110e55

closed 2020-09-25 by trinamic-LH