analogdevicesinc/PyTrinamic #26: Not possible to control two motors with cansocket?

MrGloriousFast, 2020-09-23:
Hello i successfully managed to control two motors with the default USB connection. However i struggle to understand how i would do this with a socketcan.

With the usb interface i could just open a new interface on a second port and then have one motor initialized with the first port and the second one with the second port:

```
connectionManager0 = ConnectionManager("--port=/dev/ttyACM0", debug=True)
connectionManager1 = ConnectionManager("--port=/dev/ttyACM1", debug=True)

myInterface0 = connectionManager0.connect()
myInterface1 = connectionManager1.connect()

Motor0 = TMCM_1260(myInterface0)
Motor1 = TMCM_1260(myInterface1)
```

How ever if i try to do the same with my can socket the problem is that i only have one port. the can0!

```
connectionManager0 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000", debug=True)
connectionManager1 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000", debug=True)

myInterface0 = connectionManager0.connect()
myInterface1 = connectionManager1.connect()

Motor0 = TMCM_1260(myInterface0)
Motor1 = TMCM_1260(myInterface1)
```
Both motors will do the same commands regardless if i use Motor0 or Motor1. I tried changing the module id in the hope this would mean another can adress but it just errors out:

code:

```
$ cat twoMotor_can.py
#!/usr/bin/env python3
'''
Move a motor back and forth using the TMCM1260 module
Created on 07.07.2020
@author: JM
'''

import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
import time
import asyncio

PyTrinamic.showInfo()

connectionManager0 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000 --module-id 1", debug=True)
connectionManager1 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000 --module-id 2", debug=True)

print ('lsting connections:\n', connectionManager0.listConnections())
print ('listing interfaces:\n', connectionManager0.listInterfaces())

DEFAULT_MOTOR = 2

myInterface0 = connectionManager0.connect()
myInterface1 = connectionManager1.connect()

Motor0 = TMCM_1260(myInterface0)
Motor1 = TMCM_1260(myInterface1)

print("Preparing parameters")
Motor1.setMaxAcceleration(100000)
Motor0.setMaxAcceleration(100000)

print("Rotating")
Motor0.rotate(300000)
Motor1.rotate(100000)

time.sleep(2);

print("Stopping")
Motor1.stop()
Motor0.stop()

time.sleep(1);

print ("feedback: \n", Motor0.getActualPosition(), Motor1.getActualPosition())

print("Rotating back")
Motor1.rotate(200000)
Motor0.rotate(-200000)

time.sleep(2)

print("Stopping")
Motor1.stop()
Motor0.stop()

print()

myInterface0.close()
myInterface1.close()
```

output:

```
$ python3 twoMotor_can.py
PyTrinamic - TRINAMIC's Python Technology Access Package
Splitting string: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '1']
Commandline argument list: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '1']
Parsed commandline arguments: Namespace(data_rate=[1000000], exclude=[], host_id=[2], interface=['socketcan_tmcl'], module_id=[1], port=['any'])

Connection parameters:
        Interface:  socketcan_tmcl_interface
        Port:       any
        Blacklist:  []
        Data rate:  1000000
        Host ID:    2
        Module ID:  1

Splitting string: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '2']
Commandline argument list: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '2']
Parsed commandline arguments: Namespace(data_rate=[1000000], exclude=[], host_id=[2], interface=['socketcan_tmcl'], module_id=[2], port=['any'])

Connection parameters:
        Interface:  socketcan_tmcl_interface
        Port:       any
        Blacklist:  []
        Data rate:  1000000
        Host ID:    2
        Module ID:  2

lsting connections:
 ['can0', 'can1', 'can2', 'can3', 'can4', 'can5', 'can6', 'can7']
listing interfaces:
 ['dummy_tmcl', 'pcan_tmcl', 'socketcan_tmcl', 'kvaser_tmcl', 'slcan_tmcl', 'serial_tmcl', 'uart_ic', 'usb_tmcl', 'pcan_CANopen', 'kvaser_CANopen']
Opened bus on channel can0
Opened bus on channel can0
Preparing parameters
TMCL_Request: 02,05,05,00,000186A0,33
Traceback (most recent call last):
  File "twoMotor_can.py", line 32, in <module>
    Motor1.setMaxAcceleration(100000)
  File "/usr/local/lib/python3.7/dist-packages/PyTrinamic/modules/TMCM1260/TMCM_1260.py", line 111, in setMaxAcceleration
    self.setAxisParameter(self.APs.MaxAcceleration, acceleration)
  File "/usr/local/lib/python3.7/dist-packages/PyTrinamic/modules/TMCM1260/TMCM_1260.py", line 26, in setAxisParameter
    self.connection.setAxisParameter(apType, self.__default_motor, value)
  File "/usr/local/lib/python3.7/dist-packages/PyTrinamic/connections/tmcl_interface.py", line 163, in setAxisParameter
    return self.send(TMCL_Command.SAP, commandType, axis, value, moduleID)
  File "/usr/local/lib/python3.7/dist-packages/PyTrinamic/connections/tmcl_interface.py", line 117, in send
    reply = TMCL_Reply(self._recv(self._HOST_ID, moduleID))
  File "/usr/local/lib/python3.7/dist-packages/PyTrinamic/connections/socketcan_tmcl_interface.py", line 101, in _recv
    raise ConnectionError("Recv timed out")
ConnectionError: Recv timed out

```

How do i use two or more motors on one CAN bus? 

I have two TMCM1260 and one TMCM1240.

MrGloriousFast, 2020-09-23:
I dont know if this helps but i let the same code run with can and with usb and have saved the debug output. (It is always the same output)

code:

```
 $ cat twoMotor_both_interfaces.py
#!/usr/bin/env python3

import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
import time, sys
import asyncio

USE_CAN = False

if(len(sys.argv)>1):
  print(sys.argv)
  USE_CAN = True

PyTrinamic.showInfo()

if(USE_CAN == False):

  connectionManager0 = ConnectionManager("--port=/dev/ttyACM0", debug=True)
  connectionManager1 = ConnectionManager("--port=/dev/ttyACM1", debug=True)

if(USE_CAN):

  connectionManager0 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000", debug=True)
  connectionManager1 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000", debug=True)

myInterface0 = connectionManager0.connect()
myInterface1 = connectionManager1.connect()

Motor0 = TMCM_1260(myInterface0)
Motor1 = TMCM_1260(myInterface1)

print("Preparing parameters")
Motor1.setMaxAcceleration(100000)
Motor0.setMaxAcceleration(1000000)

print("Rotating")
Motor1.rotate(-2000000)
Motor0.rotate(2000000)

time.sleep(2);

print("Stopping")
Motor1.stop()
Motor0.stop()

time.sleep(1);

print("Rotating back")
Motor1.rotate(200000)
Motor0.rotate(-200000)

time.sleep(2)

print("Stopping")
Motor1.stop()
Motor0.stop()

print()

myInterface0.close()
myInterface1.close()
```

Here are the relecant outputs from the usb interface:
```

Open port: /dev/ttyACM0
Open port: /dev/ttyACM1
Preparing parameters
TMCL_Request: 01,05,05,00,000186A0,32
TMCL_Reply:   02,01,64,05,000186A0,93
TMCL_Request: 01,05,05,00,000F4240,9C
TMCL_Reply:   02,01,64,05,000F4240,FD
Rotating
TMCL_Request: 01,05,02,00,FFE17B80,E3
TMCL_Reply:   02,01,64,05,FFE17B80,47
TMCL_Request: 01,05,02,00,001E8480,2A
TMCL_Reply:   02,01,64,05,001E8480,8E
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,00000000,6C
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,00000000,6C
Rotating back
TMCL_Request: 01,05,02,00,00030D40,58
TMCL_Reply:   02,01,64,05,00030D40,BC
TMCL_Request: 01,05,02,00,FFFCF2C0,B5
TMCL_Reply:   02,01,64,05,FFFCF2C0,19
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,00000000,6C
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,00000000,6C

Close port: /dev/ttyACM0
Close port: /dev/ttyACM1
```

and here the can output:
```

Opened bus on channel can0
Opened bus on channel can0
Preparing parameters
TMCL_Request: 01,05,05,00,000186A0,32
TMCL_Reply:   02,01,64,05,000186A0,00
TMCL_Request: 01,05,05,00,000F4240,9C
TMCL_Reply:   02,01,64,05,000186A0,00
Rotating
TMCL_Request: 01,05,02,00,FFE17B80,E3
TMCL_Reply:   02,01,64,05,000186A0,00
TMCL_Request: 01,05,02,00,001E8480,2A
TMCL_Reply:   02,01,64,05,000186A0,00
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,000F4240,00
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,000F4240,00
Rotating back
TMCL_Request: 01,05,02,00,00030D40,58
TMCL_Reply:   02,01,64,05,000F4240,00
TMCL_Request: 01,05,02,00,FFFCF2C0,B5
TMCL_Reply:   02,01,64,05,000F4240,00
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,FFE17B80,00
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,FFE17B80,00

Closing PCAN bus
Closing PCAN bus

```
the usb code will make the motors move differently but the CANSOCKET code will make them turn in unison, ignoring the different commands.

trinamic-LH, 2020-09-23:
Hello,

first of all, thank you for the detail in reporting, it helps greatly :+1:

The issue here is that there is a difference in architecture between your USB and you CAN setup - and for the former the TMCM1240 implementation is lacking.
In the USB setup you have two different buses, with CAN just one shared bus. The way this is represented in PyTrinamic is by having exactly one ConnectionInterface per bus - that is what the ConnectionManager returns when you call its `connect()` method. In all these setups TMCL is used as the protocol on the bus for communication with the modules.

TMCL does support having multiple modules on the same bus - by using the module ID. This is where the implementation is incomplete - when creating the TMCM_1260 objects, the module needs to know what module ID it has. Currently the parameter for that is missing and as a result the TMCL implementation [defaults to module ID 1](https://github.com/trinamic/PyTrinamic/blob/master/PyTrinamic/connections/tmcl_interface.py#L40). As a result, all operations done on either one of the two TMCM_1260 use module ID 1 and talk to both modules at once.

The reason USB works fine is because you do not have two modules within one bus but rather have two separate busses each using module ID 1 within their respective TMCL communication.

As a guesstimate: I think i can get this fixed tomorrow when i am back in the office, when i investigate the problem with position mode you reported in #25

trinamic-LH, 2020-09-25:
Fixed in commit 77de8faecaf60fa1a1df33aa522051a7426ebfb7

You can now pass the moduleID parameter to the constructor of the module:
```python
# connection is the connectionInterface to the shared CAN Bus used by both modules
Motor0 = TMCM_1260(connection, moduleID=1)
Motor1 = TMCM_1260(connection, moduleID=3)
```
A new version (v0.1.26) has been released with this fix included

closed 2020-09-25 by trinamic-LH