analogdevicesinc/PyTrinamic #27: two motors on one cansocket

MrGloriousFast, 2020-09-28:
Hello PyTrinamic devs!

I have seen that you implemented the moveTo command and i have verified that it works.

However i am still struggling with having two motors on one canbus. From what i understood in the code changes i tried using different module-ids. But the two motors are still behaving exactly mirrored and ignore the different module ids.

i tried using different module ids with the same code and observed the candump. I assumed to see different outputs for the candump output but they are identical.

My setup:
i have one tmcm1260 and one tmcm1240 on the same canbus. I am running the code on a raspberry pi with a mcp2515 spi-can converter but also tested the code on a jetson TX2 module (which has can build in). On both systems i am using the cansocket interface with the same code.

My code (i have two lines for connection manager. one run uses the first line, the other uses the second):

```
import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
import time

PyTrinamic.showInfo()

#connectionManager = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000 --module-id 3", debug=True)
connectionManager = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000 --module-id 4", debug=True)
myInterface = connectionManager.connect()
Module_1260 = TMCM_1260(myInterface)

try:

  #set motor parameters
  print("Preparing parameters")
  Module_1260.setMaxAcceleration(2000000000)
  Module_1260.setMaxVelocity(200000)
  Module_1260.setMaxCurrent(128)

  #set the starting position as the new zero
  Module_1260.setActualPosition(0)

  print("Rotating")
  Module_1260.rotate(300000)

  time.sleep(2);

  print("Rotating back")
  Module_1260.rotate(-20000)

  time.sleep(3)
  Module_1260.stop()

  print('moveBY')
  #test the moveBy command
  Module_1260.moveTo(0,2000000)
  time.sleep(5)
  print("Stopping")
  Module_1260.stop()

  print()

  myInterface.close()

except KeyboardInterrupt:
  print('shutting down motor and exit')
  Module_1260.stop()
  myInterface.close()
  quit()

```

My output:
```

developer@TX2: $ python3 can-motor-test.py 
PyTrinamic - TRINAMIC's Python Technology Access Package
Splitting string: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '3']
Commandline argument list: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '3']
Parsed commandline arguments: Namespace(data_rate=[1000000], exclude=[], host_id=[2], interface=['socketcan_tmcl'], module_id=[3], port=['any'])

Connection parameters:
	Interface:  socketcan_tmcl_interface
	Port:       any
	Blacklist:  []
	Data rate:  1000000
	Host ID:    2
	Module ID:  3

Opened bus on channel can0
Preparing parameters
TMCL_Request: 01,05,05,00,77359400,4B
TMCL_Reply:   02,01,64,05,77359400,00
TMCL_Request: 01,05,04,00,00030D40,5A
TMCL_Reply:   02,01,64,05,77359400,00
TMCL_Request: 01,05,06,00,00000080,8C
TMCL_Reply:   02,01,64,05,00030D40,00
TMCL_Request: 01,05,01,00,00000000,07
TMCL_Reply:   02,01,64,05,00000080,00
Rotating
TMCL_Request: 01,05,02,00,000493E0,7F
TMCL_Reply:   02,01,64,05,00000080,00
Rotating back
TMCL_Request: 01,05,02,00,FFFFB1E0,97
TMCL_Reply:   02,01,64,05,00000000,00
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,00000000,00
moveBY
TMCL_Request: 01,05,04,00,001E8480,2C
TMCL_Reply:   02,01,64,05,000493E0,00
TMCL_Request: 01,04,00,00,00000000,05
TMCL_Reply:   02,01,64,05,000493E0,00
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,FFFFB1E0,00

Closing PCAN bus
developer@TX2: $ nano can-motor-test.py 
developer@TX2: $ python3 can-motor-test.py 
PyTrinamic - TRINAMIC's Python Technology Access Package
Splitting string: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '4']
Commandline argument list: ['--interface', 'socketcan_tmcl', '--data-rate', '1000000', '--module-id', '4']
Parsed commandline arguments: Namespace(data_rate=[1000000], exclude=[], host_id=[2], interface=['socketcan_tmcl'], module_id=[4], port=['any'])

Connection parameters:
	Interface:  socketcan_tmcl_interface
	Port:       any
	Blacklist:  []
	Data rate:  1000000
	Host ID:    2
	Module ID:  4

Opened bus on channel can0
Preparing parameters
TMCL_Request: 01,05,05,00,77359400,4B
TMCL_Reply:   02,01,64,05,77359400,00
TMCL_Request: 01,05,04,00,00030D40,5A
TMCL_Reply:   02,01,64,05,77359400,00
TMCL_Request: 01,05,06,00,00000080,8C
TMCL_Reply:   02,01,64,05,00030D40,00
TMCL_Request: 01,05,01,00,00000000,07
TMCL_Reply:   02,01,64,05,00000080,00
Rotating
TMCL_Request: 01,05,02,00,000493E0,7F
TMCL_Reply:   02,01,64,05,00000080,00
Rotating back
TMCL_Request: 01,05,02,00,FFFFB1E0,97
TMCL_Reply:   02,01,64,05,00000000,00
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,000493E0,00
moveBY
TMCL_Request: 01,05,04,00,001E8480,2C
TMCL_Reply:   02,01,64,05,000493E0,00
TMCL_Request: 01,04,00,00,00000000,05
TMCL_Reply:   02,01,64,05,FFFFB1E0,00
Stopping
TMCL_Request: 01,05,02,00,00000000,08
TMCL_Reply:   02,01,64,05,FFFFB1E0,00

Closing PCAN bus

```

Just for completeness i will also attach the candump can0 output of the two runs:

```
developer@TX2: $ candump can0
  can0  001   [8]  05 05 00 77 35 94 00 4B
  can0  002   [8]  01 64 05 77 35 94 00 00
  can0  002   [8]  01 64 05 77 35 94 00 00
  can0  001   [8]  05 04 00 00 03 0D 40 5A
  can0  001   [8]  05 06 00 00 00 00 80 8C
  can0  002   [8]  01 64 05 00 03 0D 40 00
  can0  002   [8]  01 64 05 00 00 00 80 00
  can0  002   [8]  01 64 05 00 00 00 80 00
  can0  001   [8]  05 01 00 00 00 00 00 07
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  001   [8]  05 02 00 00 04 93 E0 7F
  can0  002   [8]  01 64 05 00 04 93 E0 00
  can0  002   [8]  01 64 05 00 04 93 E0 00
  can0  001   [8]  05 02 00 FF FF B1 E0 97
  can0  002   [8]  01 64 05 FF FF B1 E0 00
  can0  002   [8]  01 64 05 FF FF B1 E0 00
  can0  001   [8]  05 02 00 00 00 00 00 08
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  001   [8]  05 04 00 00 1E 84 80 2C
  can0  002   [8]  01 64 05 00 1E 84 80 00
  can0  002   [8]  01 64 05 00 1E 84 80 00
  can0  001   [8]  04 00 00 00 00 00 00 05
  can0  002   [8]  01 64 04 00 00 00 00 00
  can0  002   [8]  01 64 04 00 00 00 00 00
  can0  001   [8]  05 02 00 00 00 00 00 08
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  002   [8]  01 64 05 00 00 00 00 00
^Cdeveloper@TX2: $ candump can0
  can0  001   [8]  05 05 00 77 35 94 00 4B
  can0  002   [8]  01 64 05 77 35 94 00 00
  can0  002   [8]  01 64 05 77 35 94 00 00
  can0  001   [8]  05 04 00 00 03 0D 40 5A
  can0  001   [8]  05 06 00 00 00 00 80 8C
  can0  002   [8]  01 64 05 00 03 0D 40 00
  can0  002   [8]  01 64 05 00 00 00 80 00
  can0  002   [8]  01 64 05 00 00 00 80 00
  can0  001   [8]  05 01 00 00 00 00 00 07
  can0  001   [8]  05 02 00 00 04 93 E0 7F
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  002   [8]  01 64 05 00 04 93 E0 00
  can0  002   [8]  01 64 05 00 04 93 E0 00
  can0  001   [8]  05 02 00 FF FF B1 E0 97
  can0  002   [8]  01 64 05 FF FF B1 E0 00
  can0  002   [8]  01 64 05 FF FF B1 E0 00
  can0  001   [8]  05 02 00 00 00 00 00 08
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  001   [8]  05 04 00 00 1E 84 80 2C
  can0  002   [8]  01 64 05 00 1E 84 80 00
  can0  002   [8]  01 64 05 00 1E 84 80 00
  can0  001   [8]  04 00 00 00 00 00 00 05
  can0  002   [8]  01 64 04 00 00 00 00 00
  can0  002   [8]  01 64 04 00 00 00 00 00
  can0  001   [8]  05 02 00 00 00 00 00 08
  can0  002   [8]  01 64 05 00 00 00 00 00
  can0  002   [8]  01 64 05 00 00 00 00 00

```

You can see in my debug output that the module-id's of 3 and 4 have been passed correctly as an input argument. However in the TMCL request and reply nothing has changed. I am not sure if this is correct or not. Please excuse me if i misunderstood how its supposed to work.

I also used theTMCM1260 module even though i use a 1260 and a 1240. But i noticed no difference between using one versus the other and stuck with one. I tried the code with the other module too but no changes in behaviour.

How can i use two motors on one cansocket? Am i doing something wrong here? Wouldn't i need to get unique addresses from the motors themselves and use those? Is there an example of people using two motors on one can bus?

MrGloriousFast, 2020-09-28:
I also tried using two instances of the Connection manager with different module-id's as input arguments but the motors behave exactly the same:

here is the code i used:

#!/usr/bin/env python3

import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1260.TMCM_1260 import TMCM_1260
import time, sys

USE_CAN = True
"""
if(len(sys.argv)>1):
  print(sys.argv)
  USE_CAN = True
"""
PyTrinamic.showInfo()
"""
if(USE_CAN == False):

  connectionManager0 = ConnectionManager(debug=True)
  connectionManager1 = ConnectionManager(debug=True)
"""
if(USE_CAN):

  connectionManager0 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000 --module-id 0", debug=True)
  connectionManager1 = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000 --module-id 1", debug=True)

myInterface0 = connectionManager0.connect()
myInterface1 = connectionManager1.connect()

Motor0 = TMCM_1260(myInterface0)
Motor1 = TMCM_1260(myInterface1)

print("Preparing parameters")
Motor1.setMaxAcceleration(100000)
Motor0.setMaxAcceleration(1000000)

print("Rotating")
Motor1.rotate(-2000000)
Motor0.rotate(20000)

time.sleep(2);

print("Stopping")
Motor1.stop()
Motor0.stop()

time.sleep(1);

print("Rotating back with moveTo")
Motor1.moveTo(0)
Motor0.moveTo(0)

time.sleep(5)

print("Stopping")
Motor1.stop()
Motor0.stop()

print()

myInterface0.close()
myInterface1.close()

trinamic-LH, 2020-09-28:
Hello,

you need to set the module ID when creating the TMCM_1260 object:
```python
connectionManager = ConnectionManager("--interface socketcan_tmcl --data-rate 1000000")

myInterface = connectionManager.connect(debug_interface=True)

Motor0 = TMCM_1260(myInterface, moduleID=3)
Motor1 = TMCM_1260(myInterface, moduleID=4)
```

If you haven't already, don't forget to assign the two different module IDs to your two modules first! You can do that via the TMCL IDE or by manually using `myInterface.setGlobalParameter` in python.

MrGloriousFast, 2020-09-28:
how would i 'manually using myInterface.setGlobalParameter in python.' ?

trinamic-LH, 2020-09-28:
The Module ID used by the module should be available as a global parameter. You can set this global parameter using the setGlobalParameter function:
```python
# Helper variables for extra clarity on the setGlobalParameter arguments
bank=0
newModuleID = 3
myInterface.setGlobalParameter(TMCM_1260.GPs.serialAddress, bank, newModuleID)
```

Refer to the TMCM1260 documentation regarding this GP.
The setGlobalParameter function is defined [here](https://github.com/trinamic/PyTrinamic/blob/master/PyTrinamic/connections/tmcl_interface.py#L177)

MrGloriousFast, 2020-09-29:
I ended up using the TMCL IDE to set the global parameters. I needed to change the 'can-receive-adress' instead of the serial address. I tried using the python library but for some reason i got confused and at some point i couldn't connect to the motors anymore and needed to factory reset them (is there a python command for that?).

However the TMCM_1260.GPs.serialAddress is simply the number 66.
I needed to use the number 71. which is the 'can-receive-adress'.
If you want to improve the documentation you could add an example of using two motors with the needed steps. i think its going to be rare that a user will only need one motor. Also someone who choose to use the python library will probably made that choice because he expects a quick development cycle and ready made examples.

Thank you for pointing me into the right direction. The support of PyTrinamic is non matched and you should all be proud of yourselves. I am happy to have choosen the trinamic motors.

closed 2020-09-29 by MrGloriousFast

trinamic-LH, 2020-09-29:
Hello,

I'm glad to hear you got the module ID change working with the IDE. Apologies for mixing up the GP numbers, i did not have a CAN-based setup at hand yesterday.

Multi-Module support does indeed need some more polish - after all the module code didn't even properly support it yet before you raised the issue.

Just to confirm - does your setup now work using PyTrinamic or did you fully switch to using the TMCL-IDE?

Regards,
Lenard

MrGloriousFast, 2020-09-30:
I only used TMCL-IDE to change the module-id. After that mz whole setup
worked with PyTrinamic. I only changes the CAN receive address from 1 to 3
and 4 with the IDE.