analogdevicesinc/PyTrinamic #28: multiple access on usb port

MrGloriousFast, 2020-10-06:
Hello i want to run two motors on one USB port.

However if i do so only one motor turns and the debug is telling me a different module-ID

i set one motor to module id 3 and the other to 4.

Here is the code:
```

import PyTrinamic
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.modules.TMCM1240.TMCM_1240 import TMCM_1240
import time, sys
import asyncio

PyTrinamic.showInfo()

connectionManager = ConnectionManager(debug=True)

myInterface = connectionManager.connect(debug_interface = True)

Motor0 = TMCM_1240(myInterface, moduleID=3)
Motor1 = TMCM_1240(myInterface, moduleID=4)

print("Preparing parameters")
Motor1.setMaxAcceleration(100000)
Motor0.setMaxAcceleration(100000)

print("Rotating")
Motor1.rotate(-2000000)
Motor0.rotate(20000)

time.sleep(2);

print("Stopping")
Motor1.stop()
Motor0.stop()

time.sleep(1);

print("Rotating back with moveTo")
Motor1.moveTo(0,10000)
Motor0.moveTo(0,100000)

time.sleep(5)

print("Stopping")
Motor1.stop()
Motor0.stop()

print()

myInterface.close()
```

Here is my output:

```
$ sudo python3 twoMotor-usb.py
PyTrinamic - TRINAMIC's Python Technology Access Package
Using arguments from the command line
Commandline argument list: ['twoMotor-usb.py']
Parsed commandline arguments: Namespace(data_rate=None, exclude=[], host_id=[2], interface=['usb_tmcl'], module_id=[1], port=['any'])

Connection parameters:
        Interface:  usb_tmcl_interface
        Port:       any
        Blacklist:  []
        Data rate:  115200
        Host ID:    2
        Module ID:  1

Open port: /dev/ttyACM0
Preparing parameters
TMCL_Request: 04,05,05,00,000186A0,35
TMCL_Reply:   02,03,64,05,000186A0,95
TMCL_Request: 03,05,05,00,000186A0,34
TMCL_Reply:   02,03,64,05,000186A0,95
Rotating
TMCL_Request: 04,05,02,00,FFE17B80,E6
TMCL_Reply:   02,03,64,05,FFE17B80,49
TMCL_Request: 03,05,02,00,00004E20,78
TMCL_Reply:   02,03,64,05,00004E20,DC
Stopping
TMCL_Request: 04,05,02,00,00000000,0B
TMCL_Reply:   02,03,64,05,00000000,6E
TMCL_Request: 03,05,02,00,00000000,0A
TMCL_Reply:   02,03,64,05,00000000,6E
Rotating back with moveTo
TMCL_Request: 04,05,04,00,00002710,44
TMCL_Reply:   02,03,64,05,00002710,A5
TMCL_Request: 04,04,00,00,00000000,08
TMCL_Reply:   02,03,64,04,00000000,6D
TMCL_Request: 03,05,04,00,000186A0,33
TMCL_Reply:   02,03,64,05,000186A0,95
TMCL_Request: 03,04,00,00,00000000,07
TMCL_Reply:   02,03,64,04,00000000,6D
Stopping
TMCL_Request: 04,05,02,00,00000000,0B
TMCL_Reply:   02,03,64,05,00000000,6E
TMCL_Request: 03,05,02,00,00000000,0A
TMCL_Reply:   02,03,64,05,00000000,6E

Close port: /dev/ttyACM0
```

This is a code i wrote just to send you. I have a bigger program that is hard to share. However i receive an error that i do not get from the example code above.

Here is the output i receive:

```
MOTOR SETUP complete
ERROR getting position from motor:  0
motors on position: None 0
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/nvidia/.local/lib/python3.6/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/nvidia/moebus/mqtt-folder/motor_control.py", line 218, in _loop_check
    self.stop(motor)
  File "/home/nvidia/moebus/mqtt-folder/motor_control.py", line 237, in stop
    self.motors[motor].stop()
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/modules/TMCM1260/TMCM_1260.py", line 42, in stop
    self.rotate(0)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/modules/TMCM1260/TMCM_1260.py", line 39, in rotate
    self.setAxisParameter(self.APs.TargetVelocity, velocity)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/modules/TMCM1260/TMCM_1260.py", line 27, in setAxisParameter
    self.connection.setAxisParameter(apType, self.__default_motor, value, self.MODULE_ID)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/connections/tmcl_interface.py", line 163, in setAxisParameter
    return self.send(TMCL_Command.SAP, commandType, axis, value, moduleID)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/connections/tmcl_interface.py", line 117, in send
    reply = TMCL_Reply(self._recv(self._HOST_ID, moduleID))
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/connections/serial_tmcl_interface.py", line 69, in _recv
    return self._serial.read(9)
  File "/home/nvidia/.local/lib/python3.6/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

The error seems to be that i have two motors on the same usb hub. As far as i understand i can't receive position data. But i couldn't reproduce that bug in a standalone file. However the standalone clearly shows the wrong module id.

Is my usage of the module-id correct?

trinamic-LH, 2020-10-06:
I'm assuming you have two modules connected via USB to a USB hub?
That should result in two separate USB interfaces on the OS level (two tty's here).

You should be able to talk to both modules over two different connectionInterfaces. The moduleIDs of each connection are separate - so they can both be the default 1 (or any other valid value). The moduleID identifies modules *within* one connection. Connecting two USB devices to a PC gives you two different USB connections.

closed 2020-10-06 by MrGloriousFast

trinamic-LH, 2020-10-06:
Sidenote - it was planned to have the ConnectionManager support the multiple-modules usecase, but this has not been implemented yet. The idea behind the ConnectionManager is to have an easy way of having a script run either off hardcoded connection parameters or commandline-supplied connection parameters.

A draft on how this should work would be something like this:
In the script:
```python
connectionManager = ConnectionManager()
myInterface1 = connectionManager.connect(connection=1)
myInterface2 = connectionManager.connect(connection=2)
```

with the commandline allowing usage like this:
`myScript.py --interface usb_tmcl --port COM1 --port COM2`
to have the first connection via COM1 and the second via COM2.

This is still on the ToDo list though :/

MrGloriousFast, 2020-10-06:
Thanks for your answers. I actually had the code running with two usb ports already but assumed that a hub would be just one port from the viewpoint of the OS.

Now i can make the motors spin independently but how do i know which motor is which? Can i be sure that the motors will be always assigned the same port after each restart? 

I have the two ports:
ttyACM0
ttyACM1

trinamic-LH, 2020-10-06:
That is OS-dependant.
On windows you always keep COM-Port numbers when you are connected to the same port. No idea how it works on Linux

trinamic-LH, 2020-10-06:
Also regarding the other issue in your script:
`erial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)`
It looks like you are using threading? The TMCL-Implementation is not thread-safe. That might be the issue.

MrGloriousFast, 2020-10-06:
yes i am using threading. I use it to keep the motor moving for a set amount of time. I save the time until when i want the motor to keep spinning (for example 1 second) and in another thread i loop and check if that 1 second has passed already. Is there another way of making the motor spin for a set amount of time?

trinamic-LH, 2020-10-06:
I cannot say whether the underlying OS and Python library (Serial, python-can etc.) are thread-safe. TMCL itself works using a simple request-reply architecture. You have to wait for the reply before sending another request.

There is not really a way using PyTrinamic. Im not sure whether that's doable with a TMCL script running on the module.

MrGloriousFast, 2020-10-06:
Thank you, you helped me alot.

I can make a static usb port in linux following this guide:
https://msadowski.github.io/linux-static-port/

I also will try to write a TMCL script that lets the motor spin for a fixed amount of time (100ms).

The idea is to to control the motor remotly via the internet. I have a button and when the user keeps pressing it the motor should spin as long as the user is pressing the button. I could easily make this happen with a button-down and button-release code but the internet connection might not be reliable and a release event might get lost. that is why i want to send a periodical signal as long as the use is holding the button. So in a case of lost internet connection the motor will stop by itself.

How would i call the TMCL script from within the python library?

trinamic-LH, 2020-10-06:
I think you can have the TMCL script and the python code interact via the user variables available as global parameters.

For example, you could simply set a variable to 10 from python, decrement it by 1 every 100ms and stop as soon as it reaches 0.
Or you could use one variable as a flag - PyTrinamic sets it to 1, the TMCL script then reads it and resets it to 0 and restarts its timeout.

Bird0510, 2022-02-08:
Hi,
I am using 4 TMCL_1240 motors and connect those via USB hub to the PC.

connectionManager = ConnectionManager()
myInterface1 = connectionManager.connect(connection=1)
myInterface2 = connectionManager.connect(connection=2)

This isn´t working. Is it already implemented?

Thanks a lot, Bird

trinamic-ed, 2022-02-09:
Hi,

you can also connect directly:

my_interface = ConnectionManager("--interface usb_tmcl --port COM8").connect()
eval_board = TMC2660_eval(my_interface)

Regards,
    Enrico

Bird0510, 2022-02-09:
Nice, it works. Thanks!!!