analogdevicesinc/PyTrinamic #29: TMCM 1370 Freewheeling Mode?

camisatx, 2020-10-08:
Hey Trinamic, thanks for making this great library! I am using the [TMCM 1370 module](https://github.com/trinamic/PyTrinamic/blob/master/PyTrinamic/modules/TMCM1370/TMCM_1370.py) to interface with two PAN-Drive 1378 steppers and everything is working well. However, I noticed that the `Freewheeling mode` (number 204) is not within the [axis parameters section](https://github.com/trinamic/PyTrinamic/blob/1b2a4689a2a056fc25e608ff6f29dae07745e058/PyTrinamic/modules/TMCM1370/TMCM_1370.py#L225) of the module. Is this intentional or an accidental omission? Thanks for the help!

trinamic-LH, 2020-10-09:
Hello,

that was accidental, it has been fixed in c1d8ffdf917ee753a3eabd62ea3bcf67d6facaf5

Regards,
Lenard

closed 2020-10-09 by trinamic-LH

camisatx, 2020-10-09:
Fantastic! Thanks Lenard for the super quick update. I really appreciate it. Have a great weekend!