analogdevicesinc/PyTrinamic #35: TMCM-6110 module missing

BlackbirdB51, 2020-12-23:
I have several TMCM-6110 modules in use and would like to use them with Python. Can I use a different module (which one)?

If I'd use the TMCM-6212 module, how do I choose a different motor? The TMCM-3110 methods do have a parameter for the motor, while the TMCM-6212 methods do not, they pass the default_motor (0) to the connection.get_axis_parameter (or set).

Am I correct, that subclassing TMCM-3110 with "MOTORS = 6" would give me the right module? (maybe not the eds file)

JennRines, 2021-01-14:
I have the same issue with the 6212 python scripts.  TMCM_6212_TMCL_rotateDemo.py works well for axis 0, but 
I can't figure out how to switch to a different motor.

JennRines, 2021-01-15:
The 3110 drivers work with the 6212 and take motor # as an input

BlackbirdB51, 2021-01-22:
Thanks @JennRines, until now the module TMCM_3110 works fine for TMCM_6110. I didn't see any consequences of "MOTOR = 3". Even the other motors work.

JonaHammer, 2021-12-03:
Hi, 
we are currently working on some major changes to PyTrinamic. 
The TMCM-6110 will also be supported. 
For now, you can find an example for the Module  in the branch " feature_feature_hierarchy_v2 " (https://github.com/trinamic/PyTrinamic/tree/feature_feature_hierarchy_v2).
Please be aware that the changes may still take place. A merch with the main branch of PyTrinamic will take place in the near future. 
Kind regards 
Jonathan

closed 2021-12-06 by JonaHammer