analogdevicesinc/PyTrinamic #36: Differentiating between two motors / static usb address

MrGloriousFast, 2021-01-05:
Hello i have two motors and want to give them static usb names.

eg i want to move from:

/dev/ttyACM0
/dev/ttyACM1

to something like this:

/dev/motor_axis_x
/dev/motor_axis_y

it is imperative that i differentiate between them, a mixup might damage the hardware. Because of that i want to be sure to know which motor is connected. Any combination of unplugging and replugging including or excluding restarts in between should not interfere with the /dev/ name of the motor. For now i can assume that the same usb hub is going to be used but it would be best if i could have it work regardless of usb hubs used.

I short: i want a unique identifier for each motor.

i have motor: TMCM_1260 and TMCM_1240

I can use entries in /etc/udev/rules.d/99-usb-serial.rules to create the /dev/ entires i want. For example i am using this entry right now:

KERNEL=="ttyACM*", ATTRS{bcdDevice}=="2c01", SYMLINK+="motor_axis_x"
KERNEL=="ttyACM*", ATTRS{bcdDevice}=="2d01", SYMLINK+="motor_axis_y"

This works for now. But i don't really know how the 'bcdDevice' works and if its going to change one day. (im on linux btw,nvidia  jetson tx2)

i used the following command to see all attributes of one motor:
`$ udevadm info --name=/dev/ttyACM0 --attribute-walk`

most fields are exaclty the same to see the difference between the two motors directly i used this command:

`$ diff <(udevadm info --name=/dev/ttyACM0 --attribute-walk) <(udevadm info --name=/dev/ttyACM1 --attribute-walk)
`

It prints the following output:

```
$ diff <(udevadm info --name=/dev/ttyACM0 --attribute-walk) <(udevadm info --name=/dev/ttyACM1 --attribute-walk)
8,9c8,9
<   looking at device '/devices/3530000.xhci/usb1/1-2/1-2.4/1-2.4:1.0/tty/ttyACM0':
<     KERNEL=="ttyACM0"
---
>   looking at device '/devices/3530000.xhci/usb1/1-2/1-2.3/1-2.3:1.0/tty/ttyACM1':
>     KERNEL=="ttyACM1"
13,14c13,14
<   looking at parent device '/devices/3530000.xhci/usb1/1-2/1-2.4/1-2.4:1.0':
<     KERNELS=="1-2.4:1.0"
---
>   looking at parent device '/devices/3530000.xhci/usb1/1-2/1-2.3/1-2.3:1.0':
>     KERNELS=="1-2.3:1.0"
27,28c27,28
<   looking at parent device '/devices/3530000.xhci/usb1/1-2/1-2.4':
<     KERNELS=="1-2.4"
---
>   looking at parent device '/devices/3530000.xhci/usb1/1-2/1-2.3':
>     KERNELS=="1-2.3"
41c41
<     ATTRS{bcdDevice}=="2c01"
---
>     ATTRS{bcdDevice}=="2d01"
45,46c45,46
<     ATTRS{devnum}=="11"
<     ATTRS{devpath}=="2.4"
---
>     ATTRS{devnum}=="12"
>     ATTRS{devpath}=="2.3"

```

most of the output is from my usb hub.

First i tried giving the motors different 'serial addresses' via the TMCL-IDE. I think they are represented via the 'ATTRS{devnum}' attribute. but unplugging and replugging a motor will increment the 'ATTRS{devnum}'. therefore i can never be sure which number will correspond to which motor.

So what would be the correct way to differentiate between the motors?

MrGloriousFast, 2021-01-05:
for completeness i am adding the complete attribute walk output for each motor here:

motor 1 :
```
$ udevadm info --name=/dev/ttyACM1 --attribute-walk

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/3530000.xhci/usb1/1-2/1-2.3/1-2.3:1.0/tty/ttyACM1':
    KERNEL=="ttyACM1"
    SUBSYSTEM=="tty"
    DRIVER==""

  looking at parent device '/devices/3530000.xhci/usb1/1-2/1-2.3/1-2.3:1.0':
    KERNELS=="1-2.3:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="cdc_acm"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="02"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="01"
    ATTRS{bInterfaceSubClass}=="02"
    ATTRS{bNumEndpoints}=="01"
    ATTRS{bmCapabilities}=="2"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/3530000.xhci/usb1/1-2/1-2.3':
    KERNELS=="1-2.3"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="02"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="100mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 2"
    ATTRS{bcdDevice}=="2d01"
    ATTRS{bmAttributes}=="c0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="12"
    ATTRS{devpath}=="2.3"
    ATTRS{idProduct}=="0100"
    ATTRS{idVendor}=="2a3c"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="TrinamicMotion Ctl"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="Trinamic Stepper Device "
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{serial}=="TMCSTEP"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="64"
    ATTRS{version}==" 1.10"

  looking at parent device '/devices/3530000.xhci/usb1/1-2':
    KERNELS=="1-2"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
[... 185 lines cut ...]
    ATTRS{devnum}=="1"
    ATTRS{devpath}=="0"
    ATTRS{idProduct}=="0002"
    ATTRS{idVendor}=="1d6b"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Linux 4.9.140-tegra xhci-hcd"
    ATTRS{maxchild}=="4"
    ATTRS{product}=="xHCI Host Controller"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{serial}=="3530000.xhci"
    ATTRS{speed}=="480"
    ATTRS{urbnum}=="103"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/3530000.xhci':
    KERNELS=="3530000.xhci"
    SUBSYSTEMS=="platform"
    DRIVERS=="tegra-xusb"
    ATTRS{downgrade_usb3}=="0"
    ATTRS{driver_override}=="(null)"
    ATTRS{hsic_power}=="0"

```

trinamic-LH, 2021-01-07:
Hello and happy new year!

regarding the linux /dev configuration  - i cannot help you there.

Regarding another method of identification - if your motors are on different modules, you can simply use the [version info ](https://github.com/trinamic/PyTrinamic/blob/c1bf7b334842bd5aa92fe12ba0fd4fca0543dfbd/PyTrinamic/connections/tmcl_interface.py#L154)which includes the module type.

Other than that, you have a few options of leaving identification markers:
- You could give the modules different module IDs
- You could leave an identifier in a TMCL user variable in the global parameters (make sure to keep "Do not restore user variables" turned off - global parameter 85 bank 0 ).

closed 2021-01-07 by trinamic-LH