analogdevicesinc/PyTrinamic #37: Recv timed out

a-albano90, 2021-02-17:
When testing one of the examples on my pd57-1-1378 I get a  Recv timed out error. Is it not supported?

Lionheart1810, 2021-02-23:
Which example are you testing it with?

a-albano90, 2021-02-25:
The  TMCM_1370_TMCL_rotateDemo.py file from the examples

trinamic-LH, 2021-02-25:
Could you send us the full error/stacktrace?

a-albano90, 2021-02-25:
I get the following when I run  TMCM_1370_TMCL_rotateDemo.py on my pd57-1-1378

PyTrinamic - TRINAMIC's Python Technology Access Package
Preparing parameters
Traceback (most recent call last):
  File "./TMCM_1370_TMCL_rotateDemo.py", line 24, in <module>
    Module_1370.setMaxAcceleration(10000)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/modules/TMCM1370/TMCM_1370.py", line 111, in setMaxAcceleration
    self.setAxisParameter(self.APs.MaxAcceleration, acceleration)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/modules/TMCM1370/TMCM_1370.py", line 26, in setAxisParameter
    self.connection.setAxisParameter(apType, self.__default_motor, value)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/connections/tmcl_interface.py", line 176, in setAxisParameter
    return self.send(TMCL_Command.SAP, commandType, axis, value, moduleID)
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/connections/tmcl_interface.py", line 130, in send
    reply = TMCL_Reply.from_buffer(self._recv(self._HOST_ID, moduleID))
  File "/home/nvidia/.local/lib/python3.6/site-packages/PyTrinamic/connections/socketcan_tmcl_interface.py", line 101, in _recv
    raise ConnectionError("Recv timed out")
ConnectionError: Recv timed out

trinamic-LH, 2021-03-08:
That might just be the serial ports of PC & module not being able to communicate properly due to different baud rates. In that case, serial_tmcl might still see replies - but they're all garbage data. In that case the script would not block/timeout but it also would not properly rotate the motor.

You can doublecheck the communication settings by connecting with the TMCL-IDE.