analogdevicesinc/PyTrinamic #39: ConnectionManager --module-id argument is ignored

Lexden12, 2021-03-07:
When testing motors with various IDs using socketcan_tmcl, the ConnectionManager is always using module-id 1 even though I am setting --module-id 3 in the setting.
The arglist I am using: "--interface socketcan_tmcl --module-id 3"
The interface is being parsed properly (when the interface is not specified, it fails to start the manager), but it is always connecting to the motor with module-id 1 rather than the one I specify with --module-id.

Lexden12, 2021-03-07:
After enabling debugging, I got it to show this:
[image]
As such, it does appear that the module-id is being parsed appropriately, but the ConnectionManager is just not setting the Module ID for the channel.

trinamic-LH, 2021-03-08:
Hello,

what version of PyTrinamic are you using?

On a first glance the code appears to pass the module ID along correctly. Could you send the code you used to trigger the issue?

Regards,
Lenard

Lexden12, 2021-03-10:
Hello Lenard,

Thank you for your reply, we believe that we have found a solution, but we will have to test it.
We did not realize that the newest version of PyTrinamic added a module id parameter to the TMCM_1670 init and so that was being left at its default value of 1.

closed 2021-04-15 by trinamic-LH