analogdevicesinc/PyTrinamic #42: Set Output (SIO) Module 1160

vcalukovic, 2021-04-13:
Hi, I have a question regarding Module 1160.

I want to use my stepper motor to send a trigger pulse to another program. It was previously successfully done in TMCL-IDE using the SIO command, but I don't see how it should be done using the Module 1160 from PyTrinamic.

Any advice?
Kind regards,
Vera

vcalukovic, 2021-04-20:
@trinamic-ed and @trinamic-LH 
Unfortunately I still don't have the above working, so if you could help that would be very much appreciated.

I have looked through the different functions and modules, and am now wondering whether sending the SIO should be done through the definition __setDigitalOutput__ or **setDigitalOutputRaw** in the _tmcl_interface_ class. 
But that made me stumble upon another problem. I have tried calling these definitions from there, but regardless of which definition I try to call I get the error (through the _send() def) that the **send()** definition (which is already in that class and is used for pretty much all returns) should be implemented. 

_"NotImplementedError: The TMCL interface requires an implementation of the send() function"_

Could it be that there is something wrong with the current version of the send command? 

I'm importing the module as **from PyTrinamic.connections.tmcl_interface import tmcl_interface**
followed by **_tmcl_interface = tmcl_interface()_** to introduce the class to my code, so that the difinitions could be called as **tmcl_interface.definition(input-variables)**. (Which it currently doesn't allow)

I would very much appreciate it if you could get back to me.
Thanks in advance!
Best,
Vera

Lionheart1810, 2021-04-20:
Hi, as `tmcl_interface` ist just the base class for all TMCL-compatible interfaces, it does not implement the interface-specific functions, and is not meant to be instantiated directly. This is intended for code deduplication, and there is nothing wrong with it.
For direct communication via your desired interface, you need an instance of that interface class. E.g. for USB you need `usb_tmcl_interface`.
Using that, the `send(...)` function can be used to send a TMCL-datagram to a module and receive the corresponding reply directly, without invoking module-specific functions.
If you know all your connection parameters, you could instantiate `usb_tmcl_interface` directly. For dynamic discovery, you can use the `ConnectionManager`.
For the first one, take a look into the constructor parameters for `usb_tmcl_interface`.
For the latter, look into the example scripts for your module.
(USB just for reference, can be done with all other interface types similarly)

Using your instance, `interface.send(TMCL.COMMANDS["SIO"], port, bank, value, host_id=host, module_id=module)` can be used to set the IO directly (given `port`, `bank`, `host` as host ID, `module` as module ID).

vcalukovic, 2021-04-22:
Hi @Lionheart1810, thanks for your reply! Unfortunately, I still cannot send the triggers I want to send. If you could help a bit more that would be great.

Following your reply, I have the following for my interface setup:

```
interface = "usb_tmcl_interface"
connectionManager = ConnectionManager(interface) 
myInterface = connectionManager.connect()
Module_1160 = TMCM_1160(myInterface) # this is the stepper name, model 1160
```

and understand that in order to send a SIO, I should do that with `myInterface.send(…)`.

The input suggestions I get for this are different from the once you suggested before, and are as follows:
`myInterface.send(self= , opcode= , opType= , motor= , value= , moduleID= )`

If I understand it correctly, opcode is your command, opType = 0 (from TMCL-IDE), motor=bank = 2 (from TMCL-IDE), value is the binary trigger value 0/1 (off/on). For moduleID I am not certain what the input is supposed to be, since a TypeError is thrown when I include anything other than an integer or (integer). The error says the following: 
_TypeError: unsupported operand type(s) for &: 'list' and 'int'_ or _TypeError: unsupported operand type(s) for &: ‘str’ and ‘int’_

Leaving the moduleID empty or as an integer (I tried different values), no error is thrown, but no recognizable trigger pulse for the camera I have connected it to is sent either.

I tried playing with turning it off and on with different time intervals inbetween, but unfortunately to no avail.

```
myInterface.send(TMCL.TMCL_Command.SIO, 0, 2, 0)
time.sleep(0.15)
myInterface.send(TMCL.TMCL_Command.SIO, 0, 2, 1)
time.sleep(0.15)
myInterface.send(TMCL.TMCL_Command.SIO, 0, 2, 0)
```

What is the difference with the initial trigger command you suggested, and am I missing something still? Should I use the TMCL_Reply I obtain as input for something else still, in order to get to the desired result?

Curious what you think,
Thanks!
Vera

Lionheart1810, 2021-04-22:
For me, the following code works:

```
from PyTrinamic.connections.ConnectionManager import ConnectionManager
from PyTrinamic.TMCL import TMCL_Command
import time
con = ConnectionManager("usb_tmcl_interface").connect()
con.send(TMCL_Command.SIO, 0, 2, 1)
# GPIO 0 is logically 1
time.sleep(1)
con.send(TMCL_Command.SIO, 0, 2, 0)
# GPIO 0 is logically 0
```

Note that one some modules the polarity of GPIOs is inverted compared to voltage level.

vcalukovic, 2021-04-22:
Okay, so then I should have an inconsistency with the used moduleID+hostID and the desired target moduleID+hostID?

Is the moduleID related to the stepper motor operator (so in my case model TMCM 1160, thus TMCL module TMCM_1160), or to something else? 
How do you determine what the right target moduleID and hostID are?

Thanks!

Lionheart1810, 2021-04-22:
The default module ID is set to 1 for all modules. This is not specific to module type like TMCM-1160. So unless you changed it before by GP 66, and you are using RS485, this should not be the problem. Also, you retrieve a reply, so the module actually replied to a request. What is the `reply.status`? If it is different from 100, there might be a firmware issue.
You could also try to do

```
ConnectionManager("--interface usb_tmcl").connect(debug_interface=True)
```

in order to see the raw TMCL requests and replies sent and received via that interface. Are they the same as in the TMCL-IDE?

vcalukovic, 2021-04-23:
Taking `reply = myInterface.send(TMCL.TMCL_Command.SIO, 0, 2, toggle)`, with toggle either 1 or 0, the result of `reply.status` is 100. So that should mean its not a firmware issue, right? 
Looking at `reply.module_address` and `reply.reply_address`, this gives 1 and 2 respectively - the same as I find in the Setup>Configure Module tab of the TMCL-IDE. 
What I get in debug mode in python when running my send command is the following:
For toggle = 0:
```
TMCL_Request: 01, 0E, 00, 02, 00000000, 11
TMCL_Reply:   02, 01, 64, 0E, 00000000, 75
Out[6]: <PyTrinamic.TMCL.TMCL_Reply at 0x16bfb0f1f70>
```
For toggle = 1:
```
TMCL_Request: 01, 0E, 00, 02, 00000001, 12
TMCL_Reply:   02, 01, 64, 0E, 00000001, 76
Out[5]: <PyTrinamic.TMCL.TMCL_Reply at 0x16bfb0c79d0>
```

Which also looks the same to me. Or did you mean something else from the TMCL-IDE?

vcalukovic, 2021-05-11:
We've now checked whether the trigger signal was being sent properly using an oscilloscope. Turned out it was a communication/hierarchy problem between Python and Windows. Placement of print statements around the SIO commands fixed the problem entirely.
Thanks!

closed 2021-05-11 by vcalukovic