analogdevicesinc/PyTrinamic #43: Best Trinamic controller for a 6 axis robot using StallGuard for sensorless homing

spitzbubchen, 2021-04-14:
I primary code in python and I would like to use PyTrinamic for controlling my Stewart platform (hexapod). As the hexapod is small there is no space for limit switches so I am very interested in using StallGuard for reference homing. What Trinamic controller do people recommend? Will PyTrinamic work for this application?

trinamic-LH, 2021-04-15:
Hello,

PyTrinamic will work to control Trinamic ICs and modules via Python.

For a space-constained application, a motion controller IC with integrated MOSFETs is most likely your best choice. The [TMC5130 ](https://www.trinamic.com/products/integrated-circuits/details/tmc5130a-ta/)or [TMC5041 ](https://www.trinamic.com/products/integrated-circuits/details/tmc5041-la/)offer just that for one or two motors per IC respectively.

For Evaluation, we offer Evalboards for both ICs [1] [2]. For rapid prototyping you can use breakout boards of the ICs [3] [4].

Regards,
Lenard

[1]: [TMC5130 Eval](https://www.trinamic.com/support/eval-kits/details/tmc5130-eval/)
[2]: [TMC5041 Eval](https://www.trinamic.com/support/eval-kits/details/tmc5041-eval/)
[3]: [TMC5130 BOB](https://www.trinamic.com/support/eval-kits/details/tmc5130a-bob/)
[4]: [TMC5041 BOB](https://www.trinamic.com/support/eval-kits/details/tmc5041-bob/)

spitzbubchen, 2021-04-15:
Appreciate the info @trinamic-LH! Since it is for a 6 axis robot I was thinking something like the [TMCH-6212](https://www.trinamic.com/products/modules/details/tmcm-6212/).

One question I have is how accurate is StallGuard for reference positioning?

trinamic-LH, 2021-04-16:
StallGuard works on a fullstep resolution.
The accuracy depends on a multitude of factors - for example the used homing speed, motor temperature or noise can all impact the end result.

closed 2021-10-21 by JonaHammer