analogdevicesinc/PyTrinamic #44: Set actual position to zero

julienGautier77, 2021-04-21:
Hi
Is there a way to set the actual position to zero ?
I try :
Instruction 5 (SAP)
Type 1 (Actual Position)
motor 0
value 0
The motor move . It seems to move with a step corresponding of the postion value
I just want to do someting. like the Clear button in the TMCM-IDLE software in the widget Position Mode
Is there a way to do that ?
I'm using TMC 6110 with USB connection
Thanks in advance

trinamic-ed, 2021-04-21:
Hi, please try the following:
a) set actual position to sero (SAP Actual Position) and then set also the target position to sero
or 
b) stop the motor, go to velocity mode, set the actual position to zero, switch back to position mode

julienGautier77, 2021-04-22:
Thanks a lot
with the solution a : motor go in a way and go back. 
The solution b is working

closed 2021-04-22 by julienGautier77