analogdevicesinc/PyTrinamic #45: TMCM_1270_TMCL_rotateDemo

marek1914, 2021-06-20:
I tried to run
_PyTrinamic/PyTrinamic/examples/modules/TMCM_1270/TMCM_1270_TMCL_rotateDemo.py_
I have error at the beginning - **ConnectionManager** appears to be called incorrectly 
_AttributeError: 'ConnectionManager' object has no attribute 'get_available_interfaces'_

Are the examples from **PyTrinamic** appropriate for TMCM 1270
or only working examples are in **PyTrinamicMicro**?

Lionheart1810, 2021-06-21:
Please provide the full stack trace.

Are you working on the master branch?
The example scripts in this repository on the master branch are intended to be used on a standard computer.
There, the `ConnectionManager` does not have `get_available_interfaces`, but also no example is using it.
If you are working on a different branch, it is implemented for compatibility with `PyTrinamicMicro` - but then, the example scripts in that repository should be used.

marek1914, 2021-06-26:
I'm afraid I just realized I lack experience in GIT. I suppose the learning curve may be long and time consuming
but I want to speed it up and finally make some use of TMCM-0960 & 1270.
Could you in simple words let me know if explore of "PyTrinamic library " (its examples etc)  has any sense in my case and with above set.

Lionheart1810, 2021-06-26:
I dont think that diving deep into `PyTrinamic` library makes sense in your case.
It may just be worth to look into the module functions for TMCM-1270 in this library and take the example scripts from `PyTrinamicMicro` for reference.

Also, make sure to use the commands to clone etc. from `PyTrinamicMicro`. That way, you dont have to deal with different branches and incompatibilities.

closed 2021-06-28 by marek1914

marek1914, 2021-06-29:
Dear Leonard,

I want to avoid asking you under the "issue" topic as my problems and queries don't concern code faults but simply my misunderstanding or lack of knowledge, that's why I'm writing directly to you.

As you know I'm dealing with MotionPy board and Pandrive motor, conducting element14 roadtest.
Things (as usual with prototyping etc) don't go according to the plan. I had to cancel or postpone more ambitious tasks and spend more time on the "learning curve".
Thank you again for your support and sorry for some beginner's queries.
Here are some of my doubts I hope you can help to solve.

I would like to look again at examples accompanied to libraries.
I tried to run
.moveBy (x,x,x);
.moveTo(x,x)
I see these are implemented in both PyTrinamic and PyTrinamicMicro libraries 
but I'm afraid I misuse it trying to build scripts within PyTrinamicMicro. 
Are they indeed available and what is the right syntax?
I also tried to use direct commands using TMCL_Command.XXX some works but again I must have some problems or misunderstanding with the right syntax.
Could you suggest where I should look for the best description of the above commands.
Concerning the list of commands, I've been wondering from the beginning which are these "13 functions in Python programming language" announced in TCM 0960 papers. I supposedly looked at it or even used it but which ones are they? Where is the description?
I’m sorry again for these immature questions.
Hoping I will reach the appropriate knowledge level soon and stop disturbing you.

Greetings
Marek