analogdevicesinc/PyTrinamic #57: TMC4671 - What is the measuring unit for the Velocity and Acceleration ?

marlesus, 2022-02-08:
Hi all,

It is not clear what is the unit of measure for the Velocity and Acceleration in Closed Loop (FOC) for the TMC4671 motion controller.
It is only specified that the velocity in Open Loop is RPM (Revolutions per Minute), but there are no references for the other cases.
Can we find these units of measure somewhere else ?

Best regards,
Paul

trinamic-ed, 2022-02-09:
Hi,

the TMC4671 can use [mechanical revolutions/second] or [electrical revolutions/second] for velocity units. This depends on the register: VELOCITY_SELECTION

Please have a look to:  www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC4671-LA_datasheet_Rev2.07.pdf in chapter 4.6.11 and page 95. Selections with "phi_e_" at the beginning use electrical revolutions, and mechanical revolutions start with "phi_m_".

For e.g. a stepper motor with 50 pole pairs: 50 electrical revolutions == 1 mechanical revolution.

Regards,
    Enrico

marlesus, 2022-02-09:
Hi Enrico,

Thanks for the reply. Actually, I was looking for the answer with the RPM, and in section 4.6.11, it is stated that if VELOCITY_METER_SELECTION=0, then the displayed value for the Velocity is in RPM. I need this, as I use the phi_m angle for the Velocity Controller.

But be careful with the electrical vs. mechanical revolutions, because it is the other way around actually:
1 mechanical revolution == 50 electrical revolution (for a 50 pole pair stepper motor).
Additionally, the datasheet has some contradictions regarding this topic:
Section 4.3.2 (Page 26): phi_e = phi_m * N_POLE_PAIRS (CORRECT)
Page 85: inside description from register 0x1B: phi_e = phi_m/N_POLE_PAIRS (WRONG)
Page 87: inside description from register 0x2A: phi_e = phi_m * N_POLE_PAIRS (CORRECT)

Best regards,
Paul

trinamic-ed, 2022-02-09:
Hi,

VELOCITY_METER_SELECTION=0 is the standard. Electrical or mechanical rotations are defined by VELOCITY_SELECTION (not ..._METER_SELECTION).

Section 4.3.2 (Page 26): phi_e = phi_m * N_POLE_PAIRS (CORRECT)   => correct
Page 85: inside description from register 0x1B: phi_e = phi_m/N_POLE_PAIRS (WRONG) => yes, it is really wrong
Page 87: inside description from register 0x2A: phi_e = phi_m * N_POLE_PAIRS (CORRECT) => correct

So the velocity measured in phi_e is "number of pole pairs" * phi_m.

Thanks for the mistake in the datasheet, will forward the info to the IC team.

Regards,
Enrico

closed 2022-03-14 by trinamic