analogdevicesinc/PyTrinamic #69: TMCM6214 module

L1ikhith, 2022-09-28:
Hello, thanks for sharing this amazing library . I am looking for TMCM6214 module, with the help of other modules I tried to create a new module file (TMCM6214 ), but when I am calling using the same import function(from pytrinamic.modules import TMCM6214) its giving a error saying its not able to find the file. Can you please share the process for creating a new module file and how to integrate the new file

trinamic-bp, 2022-09-30:
Hi @L1ikhith,

Yes sorry, the TMCM-6214 is not implemented. These are the modules that are available: [https://github.com/trinamic/PyTrinamic/tree/master/pytrinamic/modules](https://github.com/trinamic/PyTrinamic/tree/master/pytrinamic/modules)
It is not that easy to create a new module, but you could use PyTrinamic for the TMCM-6214 like this
```python
from pytrinamic.connections import ConnectionManager

ap = {
    "Maximum current": 6,
    "Standby current": 7,
    "Position reached flag": 8,
    # ...
}

connection_manager = ConnectionManager()  # If no Interface is selected , the default interface is usb_tmcl
with connection_manager.connect() as interface:
    for axis in range(6):
        interface.set_axis_parameter(ap["Maximum current"], axis, 128)
    
    interface.move(..)
    interface.get_axis_parameter(ap["Position reached flag"],..)
```
You just need to fill the rest of the "Axis Parameters" you need into the ap dictionary from the modules firmware manual.

trinamic-bp, 2022-12-16:
Hi @L1ikhith,
we added the TMCM-6214 module to the Python package, also a few examples are available here on GitHub.
Can we close this issue?

closed 2023-01-04 by trinamic-bp