analogdevicesinc/PyTrinamic #72: Missing boards TMCM 0930 and TMCM 1230

PWiljes, 2022-10-20:
Hello,

can you please add the boards TMCM 0930 and TMCM 1230?

Thanks

trinamic-bp, 2023-06-14:
We finally have the TMCM-1230 in PyTrinamic.
Because TMCM-1230 and TMCM-1231 firmware is almost identical, we named the class TMCM123x_0_1, which can be uses for both modules. Under [examples/modules/TMCM123x_0_1](https://github.com/trinamic/PyTrinamic/tree/0ebf67227e16ec35699e3b81cb28c199ff03a38a/examples/modules/TMCM123x_0_1) there are examples.

trinamic-bp, 2023-06-15:
Concerning the TMCM-0930, what would be the use cases to have it in PyTrinamic? This is a control/communication module only. It is able to control TMCM-1230/TMCM-1231 modules standalone for example. With PyTrinamic there would be no need to have this module in the system.

PWiljes, 2023-06-15:
Sometimes during experiments, one needs to drive motors synchronized as
good as possible. But not in stand-alone mode, the motion needs to be
triggered by some software scan routine on a linux machine. I thought that
might be a job for the 0930 but maybe I got the idea of that module wrong.

For now, we are fine to trigger those motions with some time delay. And
for the future, we will see.

(Issue #72)

 

XLLAURANCNFSM6AAAAAARKGO2IY> .

trinamic-bp, 2023-06-16:
Tank you for your reply.
The TMCM-0930 would just send successive RS485 messages to each module. Synchronized move is not implemented at the moment. Using CAN directly would be a good alternative, as you could achieved very low delays (not sure if you do that already or using RS485). Using a multi axis board would be another, maybe even better solution.

trinamic-bp, 2024-02-08:
Closed for now, please reopen if you have anything related to this topic.

closed 2024-02-08 by trinamic-bp