analogdevicesinc/PyTrinamic #73: StallGuard2 negative Threshold retrieved unsigned

HKM-AS, 2022-12-05:
Hi, the SGT value should be retrieved signed, otherwise negative SGT values are shown incorrectly.

I am using TMCM1141 module, but I think this applies to other modules, too.

**Reproduce bug:**
```
motor.stallguard2.set_threshold(-1)
motor.stallguard2.get_threshold())
```
`4294967295`

**By modifying pytrinamic/features/stallguard2_module.py this is fixable**, the get_threshold function needs to be modified, so that get_axis_parameter is called with the argument signed=True

`return self._parent.get_axis_parameter(self._aps.SG2Threshold, self._axis, signed=True)`

trinamic-bp, 2022-12-06:
Thank you very much for telling us and creating the pull request to fix that. This is great!

closed 2022-12-06 by trinamic-bp

trinamic-bp, 2022-12-07:
I just want to mention that we need to create a new release before this change is available in PyTrinamic that is installed from PyPI. We will have this release soon, probably till the end of the year.