analogdevicesinc/PyTrinamic #75: Different Trinamic Boards have the Same identification

SamHerts, 2022-12-05:
TMCM-6214 and TMCM-1260 are identical in terms of USB identification.
On Linux, LSUSB command shows the following for TMCM-6214:

```

Bus 001 Device 003: ID 2a3c:0100 Trinamic Motion Control GmbH & Co KG Stepper Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2a3c Trinamic Motion Control GmbH & Co KG
  idProduct          0x0100 Stepper Device
  bcdDevice            2.00
  iManufacturer           1 Trinamic Motion Control
  iProduct                2 Stepper Device
  iSerial                 3 TMCSTEP
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

And TMCM1260 shows:

```

Bus 001 Device 005: ID 2a3c:0100 Trinamic Motion Control GmbH & Co KG Stepper Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2a3c Trinamic Motion Control GmbH & Co KG
  idProduct          0x0100 Stepper Device
  bcdDevice           2e.01
  iManufacturer           1 TrinamicMotion Ctl
  iProduct                2 Trinamic Stepper Device 
  iSerial                 3 TMCSTEP
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

The two key parameters are 

```
idVendor           0x2a3c Trinamic Motion Control GmbH & Co KG
  idProduct          0x0100 Stepper Device
```

Which are identical for both products. This goes against requirements and standards and makes discovery of differing devices impossible.

Is there another way to differentiate the two products, such as however TMCL-IDE is differentiating?

Issue #36 also pointed this out, but I was hoping for a better solution.

trinamic-ed, 2022-12-06:
Hi,

yes, the TMCM-6214 and TMCM-1260 are identical in terms of USB identification. For the USB identification our modules and evaluation boards use this general Trinamic VID: 0x2A3C and these product ID's:

TRINAMIC Stepper Device"            PID: 0x0100
TRINAMIC BLDC/PMSM Device"    PID: 0x0200
TRINAMIC Evaluation Device"        PID: 0x0700

Thus, it is more a category of the product type, not the product itself. To know which product is connected, the TMCL-IDE is using the getVersion command as mentioned in Issue #36. This gives us the possibility to use the same mechanism for Serial, USB, and CAN interfaces. 

Regards,
      Enrico

SamHerts, 2022-12-06:
Why is the Product Type not specified in the firmware documentation?
For both TMCM-6214 and TMCM-1260 the OPCODE 136 is only documented as returning the firmware version, not the product ID.

[image]

trinamic-bp, 2024-02-08:
Also check out this [answer to #105](https://github.com/trinamic/PyTrinamic/issues/105#issuecomment-1932435230)
The `get firmware version` indeed also returns the module code.

trinamic-bp, 2024-02-08:
Can we actually close this issue, fixing the firmware manual is beyond the scope of the PyTrinamic issue tracker?

closed 2024-07-18 by trinamic-bp