analogdevicesinc/PyTrinamic #89: Broken/Out of Date Example(s)

dscarbro21, 2023-03-24:
I'm trying to integrate my TMC5130-Eval board with a Python script to control some other devices, and have immediately run into an issue using pyTrinamic. The microstep example for the 5130 appears to be completely out of date with the current package architecture. 

First, the file has an undefined reference, `TMC5130` on line 67 - `TMCL5130_Eval` is however defined. 

Second, members from the TMC5130_Eval class that are called in the example are non-existent. `eval.showChipInfo()` and `eval.IC` are not defined. Fixing this took some time to dig out how the eval and ics are actually organized. And upon looking at another example, `rotate_demo.py` for the TMC5072, it's written in a completely different form to make use of `set_axis_parameter` which is more in-line with what's done in the TMCL-IDE anyway.

Are the examples going to be updated? The last commits on the 5130 examples were over a year ago. If not, a disclaimer in the readme saying that not all examples are up to date and functional could have saved me a couple hours, and could save others from the same trouble.

trinamic-ASU, 2023-03-27:
Thank you for bringing the out of date examples in Pytrinamic to our attention. We are very sorry for any inconvenience this may have caused you.

We wanted to let you know that we have already fixed the issue and updated the examples to ensure they are up-to-date and accurate. We value your feedback and appreciate you taking the time to bring this to our attention.

If there is anything else we can assist you with or any further feedback you have for us, please do not hesitate to reach out.

dscarbro21, 2023-03-27:
I don't see any new commits to master. Are the updated examples in a different branch?

trinamic-ASU, 2023-03-28:
To ensure that you have access to the latest changes, we recommend checking out the dev branch. This branch contains all the latest updates that have been implemented by our team.

If you encounter any issues while reviewing the dev branch, please let us know. 

Thank you for your continued support.

dscarbro21, 2023-03-28:
Ok thank you. I might suggest adding a note to the ReadMe that says that up to date examples can be found in the dev branch. The link that is in there currently under Getting Started directs to the examples folder on the master branch. So either merging the new examples to master or adjusting the link would make the end-user experience a lot better and save a lot of wasted time.

trinamic-ASU, 2023-03-29:
The latest changes have been committed to the master branch now.

Please feel free to reach out if you have any further queries.

We appreciate your continued support.

closed 2023-04-05 by dscarbro21