analogdevicesinc/PyTrinamic #91: How to enable the velocity ramp running in the TMC-Eval FW from pytrinamic ?

marlesus, 2023-06-08:
Hi there,

I am tryting to enable the velocity ramp option that is available within the Position Mode from TMCL-IDE.
Can you please tell me how can I do this using the pytrinamic ? Do you have an example of code snippet for this ?

Thank you very much for your time!

Best regards,
Paul

trinamic-bp, 2023-06-09:
Hi Paul,
Could you tell us what eval board you are using. I guess then it is easier for us to help you.

BR
Benjamin

marlesus, 2023-06-09:
Hi Benjamin,

Oh, sorry, I forgot to specify this.
I am using the TMC4671 Eval board with a BLDC motor & ABN Encoder.

Best,
Paul

trinamic-bp, 2023-06-09:
Code examples for the TMC4671-EVAL can be found under [examples/evalboards/TMC4671](https://github.com/trinamic/PyTrinamic/tree/12a0df5028a1af1bbff27efec5fb31d13bc4e72d/examples/evalboards/TMC4671). But I think the examples only cover torque mode. Not sure what the "enable velocity ramp option" in the TMCL-IDE is about, I'll take a look at it.

trinamic-bp, 2023-06-09:
I think I got it now. The firmware on the Ladungsbruecke implements this ramp. The firmware maybe does its own position control and is feeding the PID_VELOCITY_TARGET, not sure though. This functionality is configured and controlled via a higher level (parameter) interface to the Landungsbruecke . The PyTrinamic examples only operate on pure register-level. Technically PyTrinamic could also talk to the firmware on the Landungsbruecke via this higher level interface, but I'm not sure if you really want that? I guess important takeaway is that this position/velocity ramp is not implemented in the TMC4671 chip itself.

marlesus, 2023-06-09:
Hi Benjamin,

yes, that‘s exactly what I would like to achieve: enable and use the velocity ramp (which is done by simply ticking a checkbox in the Position Mode window from TMCL-IDE) from python using the PyTrinamic libraries…
So, there is no example on how to do that?
It‘s a pity… This is really a useful feature, as it limit the acceleration and avoids the unwanted bumps especially when starting a move…
But there‘s no way on how you can guide me to enable this velocity ramping from PyTrinamic ?

Best regards,
Paul

trinamic-bp, 2023-06-12:
We are trying to create you an example.
It will probably be something like this.
```py
with ConnectionManager().connect() as my_interface:

    # Create TMC4671-EVAL and TMC6100-EVAL class which communicate over the Landungsbrücke via TMCL
    mc_eval = TMC4671_eval(my_interface)
    drv_eval = TMC6100_eval(my_interface)
    
    ... # lots of mc_eval.write_register()

    my_interface.set_axis_parameter(12, 0, 1)
```
Via the `set_axis_parameter()` function we can access the "higher level parameters" I was talking about before.
But yeah, a real example is coming.

marlesus, 2023-06-12:
Hi Benjamin,

Thank you very much!
Much appreciated.

Best regards,
Paul

trinamic-bp, 2023-06-14:
Hi Paul,
we added a new example: [examples/evalboards/TMC4671/TMC4671_eval_TMC6100_eval_BLDC_ABN_encoder_position_mode_linear_ramp.py](https://github.com/trinamic/PyTrinamic/blob/0ebf67227e16ec35699e3b81cb28c199ff03a38a/examples/evalboards/TMC4671/TMC4671_eval_TMC6100_eval_BLDC_ABN_encoder_position_mode_linear_ramp.py)
I hope you can make sens of it. In line [98 to 100](https://github.com/trinamic/PyTrinamic/blob/0ebf67227e16ec35699e3b81cb28c199ff03a38a/examples/evalboards/TMC4671/TMC4671_eval_TMC6100_eval_BLDC_ABN_encoder_position_mode_linear_ramp.py#L98) the ramp settings are updated.

Please note that you will also need to update to the latest PyTrinamic version (0.2.6), we did a few updates there.

closed 2024-02-08 by trinamic-bp