analogdevicesinc/PyTrinamic #96: Missing reference search for module 6214

petertumfart, 2023-08-24:
There is no command for the reference search implemented for module 6214.

JeffPS13, 2023-11-10:
Hello, was the reference search implemented for module 6214?

trinamic-bp, 2023-11-13:
Hi. There is no TMCL command that triggers some kind of homing routine (reference search). Something like this only exists for the CANopen firmware variant. With the TMCL firmware the host system must command such a procedure. Alternatively one could implement a homing routine with a TMCL script.

trinamic-bp, 2023-11-13:
Sorry, this was not correct, the TMCM-6214 does implement the RFS command.

petertumfart, 2023-11-13:
Hi you have to call it via the interface, not via the motor (which was kind of unintuitive to me).

`intererface.reference_search(0, axis_nr)` triggers the reference search
`intererface.reference_search(1, axis_nr)` triggers the reference search

trinamic-bp, 2023-11-13:
Right, this is not very nice.

This is how it would look like in a minimal script:
```py
from pytrinamic.connections import ConnectionManager
from pytrinamic.modules import TMCM6214

connectionManager = ConnectionManager("--interface kvaser_tmcl")

with connectionManager.connect() as myInterface:
    module = TMCM6214(myInterface)

    for i, motor in enumerate(module.motors):
        motor.set_axis_parameter(motor.AP.ReferenceSearchMode, 2)  # Search right stop switch, then search left stop switch.
        module.interface.reference_search(0, i)  # Start reference search
```
We could add a per motor function like with the TMCM-3110: [TMCM3110.py#L42](https://github.com/trinamic/PyTrinamic/blob/ff1b98d7c55a57cb4f7dd4613c179abafdb2e6a8/pytrinamic/modules/TMCM3110.py#L42), to make it look better.

JeffPS13, 2023-11-13:
Thank you both for the support. It would be great to have it implemented as a per motor function.