analogdevicesinc/PyTrinamic #98: TMCL Control Command 255 - software reset

JeffPS13, 2023-11-10:
Can TMCL Control Command 255 (software reset) be sent using PyTrinamic? If so a short code snippet would be appreciated. 

Thanks - Jeff

trinamic-bp, 2023-11-13:
Coming from our regular examples, something like the following can be used to send any command:
```py
from pytrinamic.connections import ConnectionManager
from pytrinamic.modules import TMCM9999
from pytrinamic.tmcl import TMCLCommand

connection_manager = ConnectionManager("--interface kvaser_tmcl")

with connection_manager.connect() as my_interface:
    module = TMCM9999(my_interface)
    module.interface.send(TMCLCommand.X, 0, 0, 0)  # def send(self, opcode, op_type, motor, value, module_id=None):
```
Unfortunately there is no constant defined for the software reset command, but you can just replace `TMCLCommand.X` by 255.

JeffPS13, 2023-11-13:
Thanks again. This is exactly what I'm looking for.

closed 2025-01-29 by trinamic-bp