analogdevicesinc/PyTrinamic #102: type of module is "type", not TMCLModule

toqduj, 2023-12-14:
strangely enough, this fails:
'''python
import pytrinamic
from pytrinamic.modules import TMCM6214, TMCM6110
isinstance(TMCM6214, pytrinamic.modules.TMCLModule)
'''
when I check the type of TMCM6214, it reports it is type 'type'.

trinamic-bp, 2023-12-18:
Hi @toqduj, I think you want an object of that class to determine it's class type:
```py
import pytrinamic
from pytrinamic.modules import TMCM6214

tmcm_6214 = TMCM6214(None)  # Just for a demo, in a real world example one would need to give a valid "interface" instead of None.

print(isinstance(tmcm_6214, pytrinamic.modules.TMCLModule))
print(type(tmcm_6214))
```
Output:
```
True
<class 'pytrinamic.modules.TMCM6214.TMCM6214'>
```

toqduj, 2023-12-22:
ah, whoops, you're right. 

I was using it for letting people specify the board model in a configuration file, then using getattr to get that attribute from pytrinamic.modules, after which I'd wanted to test whether that really is a TMCLModule before creating an object. I'll find another way. Thanks though!

B.

closed 2023-12-22 by toqduj