analogdevicesinc/PyTrinamic #104: Feature request: Add IO.VIN enum member to read the voltage

BenediktBurger, 2024-01-16:
It would be good to read the voltage of the motor cards.

I found, that it is possible to read that voltage via `get_analog_input`, if you know the correct pin number.

I see two possible options:
1. add a `get_voltage` method (which does the correct calculation to return the voltage in Volts)
2. add a `VIN` enum member to the `IO` enum of the motor cards, see my PR #103 for one motor card.

Thanks for considering.

trinamic-bp, 2024-02-08:
It could actually be both, I accepted you PR, but the `get_voltage` would come in handy as well.
Note that the added `VIN` is not in the released PyTrinamic on PyPI, till we create a new release.

BenediktBurger, 2024-02-08:
I understand, that it will take some time till the next release.
However, I did not do it for immediate change ("8" will work as well), but for a better library.