analogdevicesinc/PyTrinamic #105: Identify Module Model using Connection Manager

JeffPS13, 2024-02-07:
Hello, 

I have two TMCL modules connected via usb and I connect to them using the code below. However, would like to programmatically identify the module model that is connected to each COM port.  I've used port_list in connection_manager.py to return the COM ports with TMCL modules attached but I don't see a means of returning the module model. 

Thanks,
Jeff 

```
            self.my_interface = ConnectionManager("--interface usb_tmcl --port COM11").connect()
            self.module = TMCM6214(self.my_interface,module_id=1)
            print( str(self.module))

            self.my_interface2 = ConnectionManager("--interface usb_tmcl --port COM9").connect()
            self.module2 = TMCM3351(self.my_interface2,module_id=2)
            print( str(self.module2))
```

trinamic-bp, 2024-02-07:
Hi @JeffPS13,
you could use the "get firmware version" command for this. The 32 bit integer you get contains the module model:
[image]

```py

from pytrinamic.tmcl import TMCLCommand

...

    get_fw_result = self.my_interface.send(TMCLCommand.GET_FIRMWARE_VERSION, 1, 0, 0, module_id=1)  # use module_id=2 for the other interface
    module_code = get_fw_result.value >> 16
    print(module_code)
```
I did not test this code, but this should be it.

JeffPS13, 2024-02-07:
Thanks for the quick response. When I run this I get: AttributeError: 'ConnectionManager' object has no attribute 'send'

trinamic-bp, 2024-02-08:
Did you remove the "`connect`"?
```py
self.my_interface = ConnectionManager("--interface usb_tmcl --port COM11")
```
vs
```py
self.my_interface = ConnectionManager("--interface usb_tmcl --port COM11").connect()
```

trinamic-bp, 2024-02-08:
Did you remove the "`connect`"
```py
self.my_interface = ConnectionManager("--interface usb_tmcl --port COM11")
```
vs
```py
self.my_interface = ConnectionManager("--interface usb_tmcl --port COM11").connect()
```

JeffPS13, 2024-02-08:
Still getting this error: AttributeError: 'ConnectionManager' object has no attribute 'send'

Here's my code:
```

from pytrinamic.connections import ConnectionManager
from pytrinamic.tmcl import TMCLCommand

my_interface = ConnectionManager("--interface usb_tmcl --port COM11 --module-id 1")

get_fw_result = my_interface.send(TMCLCommand.GET_FIRMWARE_VERSION, 1, 0, 0, module_id=1)  
module_code = get_fw_result.value >> 16
print(module_code)
```

trinamic-bp, 2024-02-09:
This should do it:
```py
from pytrinamic.connections import ConnectionManager
from pytrinamic.tmcl import TMCLCommand

my_interface = ConnectionManager("--interface usb_tmcl --port COM11 --module-id 1").connect()

get_fw_result = my_interface.send(TMCLCommand.GET_FIRMWARE_VERSION, 1, 0, 0, module_id=1)  
module_code = get_fw_result.value >> 16
print(module_code)
my_interface.close()  # properly close the COM-port
```
Note, that ` ConnectionManager("--interface usb_tmcl --port COM11 --module-id 1")` returns an instance of the connection manager class. If you take this connection manager instance and call the `connect()` function you get the an instance of the `UsbTmclInterface` class, which has the `send()` function.

BTW: To not forget the close() you better use the with statement
```py
from pytrinamic.connections import ConnectionManager
from pytrinamic.tmcl import TMCLCommand

connection_manager = ConnectionManager("--interface usb_tmcl --port COM11 --module-id 1")

with connection_manager.connect() as my_interface:
    get_fw_result = my_interface.send(TMCLCommand.GET_FIRMWARE_VERSION, 1, 0, 0, module_id=1)  
    module_code = get_fw_result.value >> 16
    print(module_code)
# my_interface.close() is automatically called when exiting the scope of the with context
```

JeffPS13, 2024-02-09:
Thanks, this works and returns the model numbers of the modules connected to each COM-port. 

Regarding using "with" and or close(), I'm a bit confused. My application is constantly sending and receiving with the modules, what is the benefit of closing and opening the interface multiple times? I close the interface when the application closes. If I'm not using best practices please advise. 

Thank you very much,
Jeff

trinamic-bp, 2024-02-09:
There is no need to close and reopen multiple times, just make sure to do the close if the application closes.

trinamic-bp, 2025-01-29:
If you encounter any further problems or have additional questions, please feel free to reopen this issue or create a new issue.

closed 2025-01-29 by trinamic-bp