analogdevicesinc/PyTrinamic #114: Reference/homing issue

kevintom1995, 2024-08-26:
The TMCM-3212 module is not detecting the home sensor when doing the reference search. I used the reference search code from the TMCM3110 module. Is this the correct code to do the reference search for this module? 
Reference #53

     def  go_to_home_position(self):
        print('Starting homing procedure')
        # Set reference speed
        self.motor.set_axis_parameter(self.motor.AP.ReferenceSearchMode, 7)
        self.motor.set_axis_parameter(self.motor.AP.ReferenceSearchSpeed, 10000)
        self.motor.set_axis_parameter(self.motor.AP.RefSwitchSpeed, 500)

        home_state=self.motor.get_axis_parameter(self.motor.AP.HomeSwitch)
        print(f'Home state: {home_state}')

        initial_status=self.get_reference_search_status(motor=0)
        print(f'Reference search status initially: {initial_status}')

        #Start reference search
        self.start_reference_search(motor=0, mode=7)
        print('Starting reference search')
        
        #waiting for reference search to complete
        self.connection.wait_reference_search(motor=0)
        print('Reference search completed')
   
        #assigning the actual position to zero
        self.motor.set_axis_parameter(self.motor.AP.ActualPosition, 0)

trinamic-ok, 2024-08-30:
The code is correct for searching the home switch. Is the switch connected to the right input? For reference search mode 7, it needs to be connected to the HOME input. The HOME input is pin 3 of the REF / HOME connector (see page 15 in the hardware manual). A home switch can be connected between HOME and GND.

s-light, 2024-12-27:
if your issue is solved please close it here in github :-)

closed 2025-01-11 by kevintom1995