analogdevicesinc/PyTrinamic #117: Import problem with new version

Jo-Leh, 2024-12-18:
With the newest version I have problems with importing the module (I call `from pytrinamic.modules import TMCM1140`). The traceback is as follows:

```
    from pytrinamic.modules import TMCM1140
/opt/hostedtoolcache/Python/3.11.11/x64/lib/python3.11/site-packages/pytrinamic/modules/__init__.py:24: in <module>
    from .TMCM1617 import TMCM1617
/opt/hostedtoolcache/Python/3.11.11/x64/lib/python3.11/site-packages/pytrinamic/modules/TMCM1617.py:10: in <module>
    from ..ic import TMC4671, TMC6200

    from .tmc_ic import TMCIc
    from .tmc_ic import RegisterApiDevice
    from .tmc_ic import RegisterGroup
    from .tmc_ic import Register
    from .tmc_ic import Field
    from .tmc_ic import Access
    from .tmc_ic import Choice
    from .MAX22216 import MAX22216
    from .TMC2100 import TMC2100
    from .TMC2130 import TMC2130
    from .TMC2160 import TMC2160
    from .TMC2208 import TMC2208
    from .TMC2209 import TMC2209
    from .TMC2224 import TMC2224
    from .TMC2225 import TMC2225
    from .TMC2240 import TMC2240
    from .TMC2300 import TMC2300
    from .TMC2590 import TMC2590
    from .TMC2660 import TMC2660
    from .TMC4361 import TMC4361
    from .TMC4671 import TMC4671
    from .TMC5031 import TMC5031
    from .TMC5041 import TMC5041
    from .TMC5062 import TMC5062
    from .TMC5072 import TMC5072
    from .TMC5130 import TMC5130
    from .TMC5160 import TMC5160
    from .TMC5240 import TMC5240
    from .TMC6100 import TMC6100
    from .TMC6140 import TMC6140
    from .TMC6200 import TMC6200
    from .TMC6300 import TMC6300
    from .TMC7300 import TMC7300
    from .TMC5272 import TMC5272
    from .TMC5271 import TMC5271
>   from .TMC9660.TMC9660 import TMC9660
E   ModuleNotFoundError: No module named 'pytrinamic.ic.TMC9660'

/opt/hostedtoolcache/Python/3.11.11/x64/lib/python3.11/site-packages/pytrinamic/ic/__init__.py:36: ModuleNotFoundError
```

trinamic-bp, 2024-12-18:
Thank you for reporting this. I see the issue.

closed 2024-12-18 by trinamic-bp

trinamic-bp, 2024-12-18:
A new version of PyTrinamic is now also published to PyPI, with the fix included.

Jo-Leh, 2024-12-18:
thanks! this is great!

commit 70758b91 (2024-12-18): Add missing __init__.py for TMC9660 directory

* fixes #117