analogdevicesinc/PyTrinamic #118: license MIT or proprietary?

s-light, 2024-12-22:
hello analogdevices team,

i just found this copyright notice in this file
https://github.com/analogdevicesinc/PyTrinamic/blob/master/pytrinamic/ic/TMC2300.py
```
################################################################################
# Copyright © 2019 TRINAMIC Motion Control GmbH & Co. KG
# (now owned by Analog Devices Inc.),
#
# Copyright © 2023 Analog Devices Inc. All Rights Reserved.
# This software is proprietary to Analog Devices, Inc. and its licensors.
################################################################################
```

but the repository [itself claims to be MIT licensed.](https://github.com/analogdevicesinc/PyTrinamic/blob/master/LICENSE)

maybe i just do not understand this *license-things* 
i would expect a MIT license block  in this file too....

why iam asking?
i thought about testing out the TMC2300 with a CircuitPython setup - 
for this i would have to write a library in the end... 
and of course it would make sense to not reinvent the wheel - for such things like registername to value mappings...
but without out a proper foss license applied to the single files it is *useless* for other foss based projects and *spare-time* projects.

(i know the main target audience for analgodevices is professionals..)

trinamic-ed, 2025-01-03:
Hello,

it is not necessary to have the MIT license in every single *.h/*.c file (only the Copyright has to be there). The license info is located here: [MIT license file](https://github.com/analogdevicesinc/PyTrinamic/blob/master/LICENSE) and is valid for all the files inside the repo. Thus, please feel free to use the registernames and mappings for projects without any limitations.

s-light, 2025-01-03:
Hello trinamic-ed,
thanks for your answer!
i will write here if i have done the library..

closed 2025-01-03 by s-light