analogdevicesinc/PyTrinamic #120: Encoder keeps ticking even when the motor is stalled/ physically blocked from moving

basitimran37, 2025-01-28:
Hello, I noted that my motor was getting stalled, something was physically blocking the move. However, the encoder ticks `motor.actual_position` would keep moving until it reached the commanded position. This is weird, as I wouldn't expect encoder reading to change in this condition. 

I was using TMCM1260's `rotate_demo.py`.

trinamic-bp, 2025-01-29:
The TMCM-1260 does not perform encoder-based closed-loop control. When reading the "Actual position," you get the position the motor is expected to be in, not its actual physical/encoder position.
You can set up StallGuard2 or the "stop on deviation" mechanism to detect and handle physical blocking.

basitimran37, 2025-01-29:
Is there a way to get actual encoder ticks instead of "expected ticks"? I understand stall detection exists, but my use-case heavily relies on the actual motor shaft position instead of the expected position.

trinamic-bp, 2025-03-18:
Did you try using axis parameter 209 - Encoder position, to get the actual motor shaft position?