analogdevicesinc/PyTrinamic #123: How do I read axis parameter 207?

easybakechef, 2025-05-07:
I am trying to read axis parameter 207. My main objective is to detect and reset a stall from stallguard, and I am following instructions from this issue: https://github.com/analogdevicesinc/PyTrinamic/issues/77

This is my code:

```
import pytrinamic
from pytrinamic.connections import ConnectionManager
from pytrinamic.evalboards import TMC2209_eval

my_interface = ConnectionManager().connect()
print(my_interface)

eval_board = TMC2209_eval(my_interface)

MOTOR = 0
AXIS_PARAM_ID = 207
axis_param_val = eval_board.get_axis_parameter(MOTOR, AXIS_PARAM_ID)
print(f'Axis Param {AXIS_PARAM_ID}: {axis_param_val}')

my_interface.close()
```

But I get the following error:
```
PS C:\Users\Contractors\Repositories\TRACE-Mx> python ..\..\Documents\scripts\pytrinamic_plaground.py
Connection: type=UsbTmclInterface port=COM3 baudrate=115200
Traceback (most recent call last):
  File "..\..\Documents\scripts\pytrinamic_plaground.py", line 12, in <module>
    axis_param_val = eval_board.get_axis_parameter(MOTOR, AXIS_PARAM_ID)
  File "C:\Users\Contractors\Repositories\TRACE-Mx\.venv\lib\site-packages\pytrinamic\evalboards\tmcl_eval.py", line 52, in get_axis_parameter
    return self._connection.get_axis_parameter(ap_type, axis, self._module_id, signed=signed)
  File "C:\Users\Contractors\Repositories\TRACE-Mx\.venv\lib\site-packages\pytrinamic\connections\tmcl_interface.py", line 206, in get_axis_parameter
    value = self._send_ap_cmd(TMCLCommand.GAP, index, axis, 0, module_id, index_bit_width).value
  File "C:\Users\Contractors\Repositories\TRACE-Mx\.venv\lib\site-packages\pytrinamic\connections\tmcl_interface.py", line 202, in _send_ap_cmd
    return self.send(cmd, tmcl_type, tmcl_motor, value, module_id)
  File "C:\Users\Contractors\Repositories\TRACE-Mx\.venv\lib\site-packages\pytrinamic\connections\tmcl_interface.py", line 135, in send
    return self.send_request(request, no_reply=no_reply)
  File "C:\Users\Contractors\Repositories\TRACE-Mx\.venv\lib\site-packages\pytrinamic\connections\tmcl_interface.py", line 114, in send_request
    raise TMCLReplyStatusError(reply)
pytrinamic.tmcl.TMCLReplyStatusError:
    TMCL_Reply:   02,01,03,06,00000000,0C
                  |  |  |  |  |        `Checksum: 12
                  |  |  |  |   `Value Unsigned: 0
                  |  |  |  |   `Value Signed: 0
                  |  |  |   `Command: 6 (GAP)
                  |  |   `Status: 3 (Wrong Type)
                  |   `Module Address: 1
                   `Reply Address: 2

PS C:\Users\Contractors\Repositories\TRACE-Mx> 
```

trinamic-bp, 2025-05-09:
Hi @easybakechef, sorry about the confusion issue `#77` is very likely not about TMC2209-EVAL and not about an EVAL board at all, but rather about some Trinamic module. The axis parameters won't help you with this. The error you get, tells you that the Landungsbruecke does not implement axis parameter.

easybakechef, 2025-05-13:
Thank you so much!

closed 2025-05-13 by easybakechef

easybakechef, 2025-05-19:
@trinamic-bp do you know how I would reset a stall with the TMC2209-EVAL/Landungsbruecke? TMCL-IDE has a button to do this, but I need to re-create the same functionality in python, and I can't tell how to reset the errors associated with the stall so that I can send more motion commands.

trinamic-bp, 2025-05-20:
[image]
To see what the TMCL-IDE is doing under the hood:

1. Enable the TMCL history log output at the bottom of the TMCL-IDE
2. Select the LB's USB port
3. Go to the settings tab
4. Enable the logging for this port
5. Open the CoolStep & StallGuard Tool
6. Klick the Reset Stall button

Now you should see in the log that "Reset Stall" does a SAP command, setting AP 181 (smarEntergy stall velocity) to 0.

In Python this would be `motor.set_axis_parameter(motor.AP.smartEnergyStallVelocity, 0)`

BTW: If you are curious on what is happening on a register level, upon `SAP 181, 0, 0`,check out the LB firmware code: [TMC-EvalSystem/boards/TMC2209_eval.c#L472](https://github.com/analogdevicesinc/TMC-EvalSystem/blob/67284056637ad8ded6c995c7406364f55f4abc32/boards/TMC2209_eval.c#L472). As the value for 181 is zero, 0x000FFFFF is written into register TCOOLTHRS.

easybakechef, 2025-05-21:
Thank you! You have saved my life lol