analogdevicesinc/PyTrinamic #124: Parameter name mismatch: `MotorControlModule.move_by(..., difference, ...)` vs. `MotorControl.move_by(..., distance, ...)`

moi90, 2025-05-22:
ABC `MotorControl` and concrete implementation `MotorControlModule` differ in the naming of the distance parameter:

```python
class MotorControl(ABC):

    @abstractmethod
    def move_by(self, distance, velocity=None):
        raise NotImplementedError()

class MotorControlModule(MotorControl):

    def move_by(self, difference, velocity=None):
        self._parent.move_by(self._axis, difference, velocity)
```

This makes Pyright unhappy. I will create a Pull Request to fix this.

trinamic-bp, 2025-05-22:
Hi @moi90, thank you for reaching out about this. There is indeed a misalignment, probably not intentional.
I don't like the name `difference` for this parameter. I would rather change it everywhere to `distance`. To throw in something new `delta` would maybe be even more descriptive and intuitive.
What do you think?

moi90, 2025-05-22:
I like `delta`. ("Move by a delta of X" sounds really intuitive). `distance` sounds more like an absolute value (that can't be negative). I'll update the pull request accordingly.

trinamic-bp, 2025-05-22:
Great, thank you very much. Let's hope we don't brake too many user code, where `difference` or `distance` is used as keyword argument. But the improvement should be worth it.

PR #126 (2025-05-22): Align difference parameter name of `move_by`
Fixes #124